particle-iot/device-os #1467: REQ: Particle Function Argument length increase from 63 to 255

technobly, 2018-01-18:
## Feature/Enhancement Request

Currently the [firmware limits the Particle.function() argument length that can be received to 63 bytes](https://docs.particle.io/reference/firmware/photon/#particle-function-).  To be more in line with publish and subscribe, an increased limit of 255 would be a great upgrade at a minimum.

This line of code appears to control the limit:
https://github.com/particle-iot/firmware/blob/7925fdaac7f04976fb9349af21c5a271c08b80e7/communication/src/protocol_defs.h#L65

Changing this to 255 resulted in the firmware receiving larger function arguments from the Particle CLI.  It would probably have to be 256 though I'm guessing since it's currently set to 64 and we claim a limitation of 63.  It would also need to be restricted to everything with more memory (aka not the Core).

technobly, 2018-01-18:
#101 is somewhat related as well.

m-mcgowan, 2018-01-18:
Gets a +1 from me.  

I was going to say how about we push these definitions down into the HAL so that it can be set on a platform by platform basis, but I feel that will hurt portability and make documentation more complex, although of course none of these insurmountable.  

Originally, the storage for functions/variables was statically allocated. Now it is dynamically allocated so in principle we could increase the length of variable and function names.

aeris-ming, 2018-02-02:
+1 Sometime the argument length is over 63 I have to cut them into frames

closed 2018-05-04 by technobly

aeris-ming, 2018-06-04:
What is the max Particle Function Argument length now?

m-mcgowan, 2018-06-04:
It is 622. Please see the referenced PR - https://github.com/particle-iot/firmware/pull/1537