particle-iot/device-os #1482: Particle.keepAlive() API does not work

technobly, 2018-01-27:
## Bug Report

### Expected Behavior

`Particle.keepAlive(30); // should ping the server every 30 seconds.`

### Observed Behavior

No activity in the log output every 30 seconds as expected.

### Steps to Reproduce

- Run the test app and observe the log output.
- Fails to work in THREADED or NON-THREADED mode
- Tested with system firmware 0.7.0-rc.4 and 0.8.0-rc.1

### Test App

```c
SerialLogHandler logHandler(LOG_LEVEL_ALL);

SYSTEM_THREAD(ENABLED);

void setup() {
    Particle.keepAlive(30); // send a ping every 30 seconds
}
```

ScruffR, 2018-02-22:
Maybe the "expected" behaviour should be that you can call `Particle.keepAlive()` anytime (prior or after connection) or at least like `Particle.variable()`, `Particle.function()` and `Particle.subscribe()` *before* up to a few seconds after the connection and still work as expected.

technobly, 2018-02-22:
Expected behavior is just that, that you can call it any time and it just sets the ping interval.  The system can keep track of the last TX/RX time and compare against the keepAlive interval (which if changed to a shorter setting may kick off a ping).

PR #1536 (2018-05-03): [Electron] Adds `keepalive_source_t` to infer when Pinger interval should be updated
### Problem

`Particle.keepAlive()` API was broken since v0.6.2-rc.2 firmware on Electron where the System could override a User set ping interval.  This required a workaround of updating the keepAlive after the System made a connection to the Cloud.  See issue #1482.

### Solution

Adds `keepalive_source_t` to infer when Pinger interval should be updated.  The only time the interval will not be updated is when the User has previously set it, and the System tries to override it.  All other cases are updatable. [... cut]
closed 2018-05-03 by technobly