particle-iot/device-os #1484: usr thread not run with system firmware 0.7.0.rc6 and 0.8.0.rc1

aeris-ming, 2018-01-30:
// Choose external antenna for WiFi
STARTUP(WiFi.selectAntenna(ANT_EXTERNAL));
// Run application thread separately from Wifi Thread
SYSTEM_MODE(SEMI_AUTOMATIC);
SYSTEM_THREAD(ENABLED);

void setup()
{
        /*DO NOT adjust init sequence*/
        Serial.begin(9600);
        Serial.end();
        Serial.begin(9600);
}

void loop()
{
    Serial.printf("fuck photon...\r\n");
    Serial.println("fuck photon...");
    Serial.flush();
}

aeris-ming, 2018-01-30:
@m-mcgowan Thanks,I will
But the system firmware it is really ugly!Full of endless loop and bugs~!
It makes me very mad especially our user's device not work and we have thousands device in our warehouse.

m-mcgowan, 2018-01-30:
Let's keep this issue open - than you for bringing the problem to our attention. We will investigate and see if we can find the cause and any possible workaround.

aeris-ming, 2018-01-30:
@m-mcgowan 
1. I am using Linux
2.This usb serial port always block or not respond on Linux and Windows,that's why I write a program to flash Photon by ST-link
3.I have an external antenna
4.I will remove re-opening code and test again
5.I do not want to try the latest system firmware,actually I am still using version 0.5.4.
I just want to know if the user thread still will be block when wifi turn off or signal is weak or internet is not stable
6.We have already use RT-OS and we run system and usr code in different thread,Why usr thread be blocked?
7.I really need some help to solve this problem too
             https://github.com/particle-iot/firmware/issues/1407
8.Thanks very much!

aeris-ming, 2018-01-30:
1.I use the same usr code with system firmware 0.5.4,It works
2.I do not call Particle.connect(), but the led blink blue not white
3.I remove the re-open code and test again,it not work
4.cmd 'v' and 'i' works,no other output
[image]

m-mcgowan, 2018-01-30:
When the LED is blinking blue then the device is in [listening mode](https://docs.particle.io/guide/getting-started/modes/photon/#listening-mode). 

This will happen when you ask the device to via Particle.connect(), WiFi.connect(), or when in AUTOMATIC mode, and there are no Wi-Fi credentials configured.

aeris-ming, 2018-01-30:
1.SYSTEM_MODE(SEMI_AUTOMATIC);
2.I do not call any Particle and Wi-Fi function
3.When use version 0.5.4 it is blink white

m-mcgowan, 2018-01-30:
Which version is it not blinking white?

aeris-ming, 2018-01-30:
I have try 0.8.0-rc.1 and 0.7.0-rc.6 both blink blue

m-mcgowan, 2018-01-30:
My guess here is that when you upgraded 0.7.0/0.8.0 you did not upgrade the bootloader.  This will cause the device to enter safe mode, since it is a required dependency. In safe mode, the device always connects to the cloud - like a `SYSTEM_MODE(AUTOMATIC)` application. When the device doesn't have any Wi-Fi credentials configured, it will instead enter listening mode. 

You have two options:
1. use `particle flash --serial` to update the bootloader (download from the GitHub release page)
2. connect your device to a Wi-Fi AP and let the cloud update the bootloader.

When the device is in safe mode, the application is not executed at all. This is by design, since to do so would most likely crash as the system is missing key modules the application depends upon.

aeris-ming, 2018-01-31:
I must update bootloader If I want to update new system firmware to my customer's devices?

aeris-ming, 2018-01-31:
@m-mcgowan It works,thanks!

closed 2018-01-31 by aeris-ming