particle-iot/device-os #1486: Usr code be blocked for a very long time even ENABLE Muti-Thread!!!!

aeris-ming, 2018-02-01:
I think WIFI or Internet operation block the user code

I have test system firmware 0.5.3 0.5.4 0.6.1 0.7.0.rc.6 0.8.0.rc.1

SYSTEM_MODE(SEMI_AUTOMATIC);
SYSTEM_THREAD(ENABLED);

Just turn off your router or use wrong wifi psw or take Photon out of signal range of router!

m-mcgowan, 2018-03-14:
We need to know more details about your application code in order to investigate. Are you calling any of the system functions that will block the application thread? Here's the details: https://docs.particle.io/reference/firmware/photon/#system-functions

aeris-ming, 2018-03-16:
Sorry,I called WiFi.hasCredentials() WiFi.listening() WiFi.ready() Particle.connected().
I should debug user code carefully.
After I remove them,User thread not be block anymore.
I think when someone who reads these function name will think that these function just check a flag then return.
Maybe change the name of these function will be a good idea for example WiFi.hasCredentialsMayBlockForEver() and WiFi.readyReturnImmediately.
Does WiFi.listening() WiFi.ready() Particle.connected() just check one flag??? I call them very often.
Thanks!

aeris-ming, 2018-03-19:
@m-mcgowan I have one device before it fall into sleep,It will try to send some data to servers.
According to the Reference Publish may block for at most 60sec,It is not very bad.
https://docs.particle.io/reference/firmware/photon/#particle-publish-
ttl (time to live, 0–16777215 seconds, default 60) !! NOTE: The user-specified ttl value is not yet implemented, so changing this property will not currently have any impact.

Generally I should check If Credentials has been set up first before call Particle.connect().
But because of WiFi.hasCredentials() may block forever, I need to know what will happen if I call Particle.connect() without credentials.

/*before sleep try to connect to server*/
if (!Particle.connected()) {
        if (WiFi.hasCredentials()) {
                Particle.connect();
                /*Wait for 20sec most*/
                for (int32_t i = 0; i < 100; i++) {
                        delay(20);
                        if (Particle.connected()) {
                                break;
                        }
                }
        }
}
if (Particle.connected()) {
        /*try to publish data*/
        Particle.publish("new-data", "128degree", 10000);
}
/*fall back to sleep to save power*/
System.sleep();

m-mcgowan, 2018-05-02:
The TTL in `Particle.publish()` specifies how long the event should live for in the Cloud. It's a future-facing feature that is not yet implemented - it doesn't have any bearing on how long the function call will block for.

When the device is not connected to the cloud, then the call to Publish will simply return. I don't believe you need all the tests - but happy to be shown otherwise :-)

aeris-ming, 2018-05-03:
@m-mcgowan Thanks

closed 2018-05-03 by aeris-ming