particle-iot/device-os #1490: Bad WiFi reception causes WiFi.RSSI() to return 2 and a following WiFi.disconnect() to SOS with stack overflow

elcojacobs, 2018-02-24:
<details>
 <summary><i>
In an attempt to investigate the terrible WiFi reliability and TCP reliability that has gone unfixed for at least 6 months now, I wrote some test code and stumbled upon very bad behavior when handling WiFi.

Please stop ignoring this Particle. Read the full bug report below, because it is complex.
It is affecting my customers. I bought a lot of units from you, cores, photons, P1's. As a long time customer, I'm very disappointed that a WiFi IOT product does not take bug reports about bad WiFi reliability more seriously.
</i></summary>
</details>

## Bug Report

### Expected Behavior

When WiFi connection is lost due to bad reception:
- LED stops breathing green
- WiFi.ready() returns false
- WiFi is restarted

### Observed Behavior

To test why TCP stopped working sometimes, I wrote 2 pieces of test code:
- A python script to show what is printed over serial and to send a TCP message every second and to print TCP replies (tic-toc)
- A Photon app to reply to TCP requests and report system status.

#### The python script
- Prints anything received over serial to terminal in GREEN text
- Prints a message in RED when it sends a TCP message to the photon
- Prints a message in BLUE when it receives are TCP message from the photon
- Sends a TCP message every second

#### The firmware on the photon
- Starts a TCP server on port 6666
- Listens for clients
- Runs a system thread
- Handles TCP server start/stop with network event handlers setting a flag (relying on WiFi.ready() seems to be unreliable)
- Prints to serial: WiFi.ready(), IP address, RSSI, TCP client connected, millis()
- Prints DEBUG logs
- Prints SerialLogHandler traceLog(LOG_LEVEL_TRACE);

#### Behavior
Before adding `if(signal == 2){` block 
- TCP tic/toc just seems to stop at some point and doesn't recover
- WiFi.RSSI returns 2 when problems start. This is triggered by bad reception. Doesn't recover after reception restored.

In an attempt to fix I added the block to handle RSSI of 2.
- When the code block is triggered, WiFi.disconnect() causes an SOS with 13 blinks: stack overflow.
- Device never returns from SOS, stays in SOS until hard reset.

Output of python script:
```
py->hw: tic
0000028203 [wiring] spark_wiring_tcpclient.cpp:144, TCPClient::available(): TRACE: recv(=1)
0000028204 [hal] socket_hal.cpp:999, socket_send(): TRACE: Write 3 bytes to socket 536909128 result=0
hw->py: toc (3 bytes sent)
py->hw: t
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -79               TCP client connected: 1         millis(): 29124
0000029214 [hal] socket_hal.cpp:778, read_packet_and_dispose(): TRACE: Socket 0 receive bytes 1 of 1
0000029214 [wiring] spark_wiring_tcpclient.cpp:144, TCPClient::available(): TRACE: recv(=1)
py<-hw: toc
0000029215 [hal] socket_hal.cpp:999, socket_send(): TRACE: Write 3 bytes to socket 536909128 result=0
hw->py: toc (3 bytes sent)
py->hw: t
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -79               TCP client connected: 1         millis(): 30125
0000030211 [hal] socket_hal.cpp:778, read_packet_and_dispose(): TRACE: Socket 0 receive bytes 1 of 1
0000030211 [wiring] spark_wiring_tcpclient.cpp:144, TCPClient::available(): TRACE: recv(=1)
0000030212 [hal] socket_hal.cpp:999, socket_send(): TRACE: Write 3 bytes to socket 536909128 result=0
hw->py: toc (3 bytes sent)
py->hw: t
py->hw: tic
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -80               TCP client connected: 1         millis(): 31126
[... 621 lines cut ...]

*Python*
```
pip3 install colorama
pip3 install pyserial
python3 wifi-test.py 
```
Edit python script to use the right IP address and serial port. Optionally change the LED pin to see whether the loop is still running.

*How to get WiFi reception bad enough to trigger behavior:*
I set up a router far enough to have wifi reception around -75dBm. Then I got a cast iron pot. When I put the photon inside the cast iron pot, the WiFi reception drops enough to trigger RSSI=2.

### Test App
On GitHub here:

https://github.com/BrewPi/firmware/tree/feature%2Fwifi-channel-switch-test

### References

https://github.com/particle-iot/firmware/issues/1371
https://community.particle.io/t/wifi-channel-switching-stops-wifi-connectivity-and-is-only-recoverable-by-reboot-bug/38275/20

---

Please stop ignoring these WiFi issues and make TCP/WiFi reliable. My product is starting to look bad because of issues with your software framework. I have reported a bug about WiFi.ready() returning true when it shouldn't 6 months ago and it just isn't handled. I have wasted at least 50 hours trying to work around these issues, while it should be handled by Particle. I have provided you extensive test code. Please at least assign someone to run it and report back. ASAP. There's not much more I can do to help.

elcojacobs, 2018-02-25:
Here is another interesting tracelog, triggered by bad reception:
It is trying to close a socket 2 times per mililsecond.

```
0000184738 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184738 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184739 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184739 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184740 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184740 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184741 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184741 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184742 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184742 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184743 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184743 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184744 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184744 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184745 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184745 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184746 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184746 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184747 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184747 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184748 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184748 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184749 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184749 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184750 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184750 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184751 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184751 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184752 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184752 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184753 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184753 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184754 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184754 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184755 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184755 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184756 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184756 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184757 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184757 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184758 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184758 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184759 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184759 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184760 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184760 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184761 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184761 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184762 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184762 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184763 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184763 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184764 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
0000184764 [wiring] spark_wiring_tcpclient.cpp:189, TCPClient::stop(): TRACE: sock -1 closesocket
[... cut ...]
hw->py: toc (3 bytes sent)
py->hw: t
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -75               TCP client connected: 1         millis(): 188254
0000189252 [hal] socket_hal.cpp:778, read_packet_and_dispose(): TRACE: Socket 0 receive bytes 1 of 1
0000189253 [wiring] spark_wiring_tcpclient.cpp:144, TCPClient::available(): TRACE: recv(=1)
py<-hw: toc
0000189253 [hal] socket_hal.cpp:999, socket_send(): TRACE: Write 3 bytes to socket 536908984 result=0
hw->py: toc (3 bytes sent)
```

elcojacobs, 2018-02-25:
Bad WiFi reception test setup:
[image]

I'm trying to break before the SOS with a hardware debugger. If I break on the serial print after the check for RSSI == 2, the breakpoint is hit, but I still get an SOS.
I suspect this is because the SOS happens in the system thread and st-util doesn't support multithreading. Will try to debug with openocd instead. If you think my code can cause the stack overflow, please point out how. I don't see it. I think it is deep inside the system framework.

elcojacobs, 2018-02-25:
This stopped the SOS (avoiding a bug I think, not fixing it):

https://github.com/BrewPi/firmware/compare/63f95d3b10d7...2b849b3184db

My code with many workarounds for handling TCP and WiFi issues can now quite reliably hold a connection (by restarting it often and handling disconnects).

Still very much worth an investigation. Please try to run the code (firmware and python) and mess with the wifi signal. It will give you a ton of information on how the firmware is handling it.

elcojacobs, 2018-02-25:
Tested on 0.8.0-rc.2:

Just rebooting the router so WiFi goes down will trigger the RSSI == 2 state, now with IP 0.0.0.0.

Seems to take longer to recover and get back online than on 0.7.0-rc.6.
Python script output below:

<details>
<pre>
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -69               TCP client connected: 1         millis(): 63213
py->hw: tic
py<-hw: toc
0000064032 [wiring] spark_wiring_tcpclient.cpp:163, TCPClient::available(): TRACE: recv(=1)
hw->py: toc (3 bytes sent)
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -70               TCP client connected: 1         millis(): 64214
py->hw: tic
py<-hw: toc
0000065032 [wiring] spark_wiring_tcpclient.cpp:163, TCPClient::available(): TRACE: recv(=1)
hw->py: toc (3 bytes sent)
py->hw: tic
py<-hw: toc
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -70               TCP client connected: 1         millis(): 65215
0000066032 [wiring] spark_wiring_tcpclient.cpp:163, TCPClient::available(): TRACE: recv(=1)
hw->py: toc (3 bytes sent)
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -69               TCP client connected: 1         millis(): 66216
py->hw: tic
0000067032 [wiring] spark_wiring_tcpclient.cpp:163, TCPClient::available(): TRACE: recv(=1)
hw->py: toc (3 bytes sent)
py<-hw: toc
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -69               TCP client connected: 1         millis(): 67217
py->hw: tic
py<-hw: toc
0000068032 [wiring] spark_wiring_tcpclient.cpp:163, TCPClient::available(): TRACE: recv(=1)
hw->py: toc (3 bytes sent)
py->hw: tic
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -70               TCP client connected: 1         millis(): 68218
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -69               TCP client connected: 1         millis(): 69219
network event 10
network event 11
py->hw: tic
Stopping TCP
TCP server stopped
WiFi.ready(): 0         IP: 0.0.0.0             RSSI: 2         TCP client connected: 0         millis(): 70220
py->hw: tic

WiFi is in ERROR state (RSSI ==2)

Stopping TCP
TCP server stopped
py->hw: tic
WiFi.ready(): 0         IP: 0.0.0.0             RSSI: 2         TCP client connected: 0         millis(): 71221

WiFi is in ERROR state (RSSI ==2)

py->hw: tic
Stopping TCP
TCP server stopped
0000071394 [hal.wlan] TRACE: connect cancel
py->hw: tic
network event 2
network event 3
network event 4
py->hw: tic
0000072120 [hal.wlan] INFO: Using internal antenna
network event 5
network event 6
py->hw: tic
0000072166 [hal.wlan] INFO: Joining brewpi-subnet
0000072166 [hal.wlan] TRACE: Free RAM connect: 46648
py->hw: tic
[... 374 lines cut ...]
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -72               TCP client connected: 1         millis(): 139289
py<-hw: toc
0000139798 [wiring] spark_wiring_tcpclient.cpp:163, TCPClient::available(): TRACE: recv(=1)
hw->py: toc (3 bytes sent)
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -72               TCP client connected: 1         millis(): 140290
0000140794 [wiring] spark_wiring_tcpclient.cpp:163, TCPClient::available(): TRACE: recv(=1)
hw->py: toc (3 bytes sent)
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -73               TCP client connected: 1         millis(): 141291
py->hw: tic
py<-hw: toc
0000141794 [wiring] spark_wiring_tcpclient.cpp:163, TCPClient::available(): TRACE: recv(=1)
hw->py: toc (3 bytes sent)
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -72               TCP client connected: 1         millis(): 142292
0000142793 [wiring] spark_wiring_tcpclient.cpp:163, TCPClient::available(): TRACE: recv(=1)
hw->py: toc (3 bytes sent)
py->hw: tic
WiFi.ready(): 1         IP: 192.168.1.61                RSSI: -73               TCP client connected: 1         millis(): 143293
py<-hw: toc
0000143794 [wiring] spark_wiring_tcpclient.cpp:163, TCPClient::available(): TRACE: recv(=1)
py->hw: tic
hw->py: toc (3 bytes sent)
</pre>
</details>

Please run the scripts yourself. It is easier with colored output.

elcojacobs, 2018-02-26:
@avtolstoy 
With the fix/photon-tcp branch, this doesn't seem to cause the stack overflow anymore:

```cpp
if(signal == 2){
            Serial.print("\n\n\nWiFi is in ERROR state (RSSI ==2)\n\n\n");

            /* This code causes a buffer overflow and the system does not recover from the resulting SOS. Don't use.
             * Leaving it here for particle to investigate. */
              WiFi.disconnect();
              WiFi.connect(WIFI_CONNECT_SKIP_LISTEN);
    }
}
```

It never connects to WiFi, but this can be expected because it is being disconnected while still trying to.

aeris-ming, 2018-03-07:
I have the same issue.
1.I use another PCB cover antenna of my device
2.power on my device,LED blink red

closed 2018-04-05 by technobly