particle-iot/device-os #1491: Support for an optional cloud connection (stop resetting a local WiFi connection that is working perfectly fine, but doesn't have Internet).

elcojacobs, 2018-02-25:
<details>
<summary><i>
When the application is running with a cloud connection and loses cloud connection, it will reset the entire WiFi connection and keep doing so until the cloud connection is restored.
This is undesirable for applications that mainly use the local network. I think the two should be handled separately, or at least the user should have a choice in this.
</i></summary>
## Bug Report

### Desired Behavior
- Photon starts up WiFi and starts breathing green.
- Photon tries to connect to cloud (alternating between green/cyan)

- When successfully connected to cloud, photon breathes cyan
- When cloud connection fails, photon reverts back to breathing green (optionally retrying without interfering with the wifi connection)

### Observed Behavior
An application that is working perfectly with WiFi.connect() on a local network without Internet is unusable when Particle.connect() is used instead.
The entire WiFi stack is reset, even though nothing is wrong with the WiFi connection. The router just can't connect to Particle.

(firmware version 0.7.0-rc.6)

It would be nice to have a cloud connection if possible (for firmware updates for example), but to have a working app if it isn't.

### References

---

avtolstoy, 2018-02-28:
Have you tried disabling `SYSTEM_FLAG_RESET_NETWORK_ON_CLOUD_ERRORS`? (see https://docs.particle.io/reference/firmware/core/#system-flags-disable-)

elcojacobs, 2018-03-04:
That seems to work just fine. I didn't know it existed. Perhaps add a reference to it in de documentation for Particle.connect() explaining that resetting the WiFi stack is the default behavior and this can be changed by setting the flag.

closed 2018-03-04 by elcojacobs