particle-iot/device-os #1493: Binary Publish Payload

towynlin, 2018-03-01:
## Feature/Enhancement Request

As a user creating a Particle-powered product
In order to minimize data usage and to lower the barrier to using a binary serialization format of my choice, e.g., CBOR
I want the option with `Particle.publish` for the `data` field to be binary, including null bytes and bytes with the high bit set,
even if the firmware API is harder to use (e.g., sending buffer length argument),
and even if the SSE output is ugly.

### References

https://community.particle.io/t/cbor-for-coap-payload-format/39614/3?u=zachary

aeris-ming, 2018-03-21:
After I know convert String to uint8_t array It is not safe.
I want the parameter of Particle.function to support binary data too.

aeris-ming, 2018-03-21:
Actually If the data will not be truncated if there is a ‘\0’ in middle.
No user API need to be change.
String Class supoort toCharArray and length.

aeris-ming, 2018-06-04:
I will try to do it by my own

aeris-ming, 2018-06-05:
```
commit e1a3cb6197428bcbe62c9bfbf59ec07525cca2a0
Author: ChangmingTian <ming@aeris.io>
Date:   Tue Jun 5 10:56:08 2018 +0800

    add binary payload feature to Particle.function

diff --git a/communication/src/functions.h b/communication/src/functions.h
index 4a5949f..ed1166c 100644
--- a/communication/src/functions.h
+++ b/communication/src/functions.h
@@ -107,7 +107,7 @@ public:
         // call the given user function
         auto callback = [=,&channel] (const void* result, SparkReturnType::Enum resultType )
             { return this->function_result(channel, result, resultType, token); };
-        call_function(function_key, function_arg, callback, NULL);
+        call_function(function_key, function_arg, callback, &function_arg_length);
         return NO_ERROR;
     }
 };
diff --git a/system/src/system_cloud_internal.cpp b/system/src/system_cloud_internal.cpp
index 19a077f..8209a3a 100644
--- a/system/src/system_cloud_internal.cpp
+++ b/system/src/system_cloud_internal.cpp
@@ -177,8 +177,9 @@ User_Func_Lookup_Table_t* find_func_by_key_or_add(const char* funcKey, const clo
 
 int call_raw_user_function(void* data, const char* param, void* reserved)
 {
+       size_t *param_length = (size_t *)reserved;
     user_function_int_str_t* fn = (user_function_int_str_t*)(data);
-    String p(param);
+    String p(param, (*param_length));
     return (*fn)(p);
 }
 
@@ -1249,11 +1250,13 @@ const void *getUserVar(const char *varKey)
     return result;
 }
 
-void userFuncScheduleImpl(User_Func_Lookup_Table_t* item, const char* paramString, bool freeParamString, SparkDescriptor::FunctionResultCallback callback)
+void userFuncScheduleImpl(User_Func_Lookup_Table_t* item, const char* paramString, bool freeParamString, SparkDescriptor::FunctionResultCallback callback, size_t *param_length)
 {
-    int result = item->pUserFunc(item->pUserFuncData, paramString, NULL);
-    if (freeParamString)
+    int result = item->pUserFunc(item->pUserFuncData, paramString, &param_length);
+    if (freeParamString) {
         delete paramString;
+       delete param_length;
+    }
     // run the cloud return on the system thread again
     SYSTEM_THREAD_CONTEXT_ASYNC(callback((const void*)result, SparkReturnType::INT));
     callback((const void*)long(result), SparkReturnType::INT);
@@ -1261,17 +1264,25 @@ void userFuncScheduleImpl(User_Func_Lookup_Table_t* item, const char* paramStrin
 
 int userFuncSchedule(const char *funcKey, const char *paramString, SparkDescriptor::FunctionResultCallback callback, void* reserved)
 {
+       size_t function_arg_length = *((size_t *)reserved);
+
     // for now, we invoke the function directly and return the result via the callback
     User_Func_Lookup_Table_t* item = find_func_by_key(funcKey);
     if (!item)
         return -1;
 
 #if PLATFORM_THREADING
-    paramString = strdup(paramString);      // ensure we have a copy since the oriignal isn't guaranteed to be available once this function returns.
-    APPLICATION_THREAD_CONTEXT_ASYNC_RESULT(userFuncScheduleImpl(item, paramString, true, callback), 0);
-    userFuncScheduleImpl(item, paramString, true, callback);
+    // ensure we have a copy since the oriignal isn't guaranteed to be available once this function returns.
+    char *string = new char[function_arg_length + 1];
+    memcpy(string, paramString, function_arg_length + 1);
+    paramString = string;
+    size_t *length = new size_t;
+    *length = function_arg_length;
+
+    APPLICATION_THREAD_CONTEXT_ASYNC_RESULT(userFuncScheduleImpl(item, paramString, true, callback, length), 0);
+    userFuncScheduleImpl(item, paramString, true, callback, length);
 #else
-    userFuncScheduleImpl(item, paramString, false, callback);
+    userFuncScheduleImpl(item, paramString, false, &function_arg_length);
 #endif
     return 0;
 }
```

aeris-ming, 2018-06-05:
@towynlin I have start my work on binary payload feature.I need some information about how to debug system firmware.Can you help me ?1.I want to call communication/src/protocol.h handle_received_message() function to trigger a callback Particle.function to do some test,Is there a document or example show how to call this function?2.How to send debug information?The same as user firmware?

aeris-ming, 2018-06-05:
```
commit 697c0bbfa60d316573228f35d15d97d0b4e01311
Author: ChangmingTian <ming@aeris.io>
Date:   Tue Jun 5 15:25:38 2018 +0800

    add binary payload feature to Particle.publish

diff --git a/communication/src/messages.cpp b/communication/src/messages.cpp
index a27938d..911a5a3 100644
--- a/communication/src/messages.cpp
+++ b/communication/src/messages.cpp
@@ -330,6 +330,40 @@ size_t Messages::event(uint8_t buf[], uint16_t message_id, const char *event_nam
   return p - buf;
 }
 
+size_t Messages::event(uint8_t buf[], uint16_t message_id, const char *event_name,
+             const char *data, size_t data_length, int ttl, EventType::Enum event_type, bool confirmable)
+{
+  uint8_t *p = buf;
+  *p++ = confirmable ? 0x40 : 0x50; // non-confirmable /confirmable, no token
+  *p++ = 0x02; // code 0.02 POST request
+  *p++ = message_id >> 8;
+  *p++ = message_id & 0xff;
+  *p++ = 0xb1; // one-byte Uri-Path option
+  *p++ = event_type;
+
+  size_t name_data_len = strnlen(event_name, MAX_EVENT_NAME_LENGTH);
+  p += event_name_uri_path(p, event_name, name_data_len);
+
+  if (60 != ttl)
+  {
+    *p++ = 0x33;
+    *p++ = (ttl >> 16) & 0xff;
+    *p++ = (ttl >> 8) & 0xff;
+    *p++ = ttl & 0xff;
+  }
+
+  if (NULL != data)
+  {
+    name_data_len = data_length > MAX_EVENT_DATA_LENGTH ? MAX_EVENT_DATA_LENGTH : data_length;
+
+    *p++ = 0xff;
+    memcpy(p, data, name_data_len);
+    p += name_data_len;
+  }
+
+  return p - buf;
+}
+
 size_t Messages::coded_ack(uint8_t* buf, uint8_t token, uint8_t code,
                            uint8_t message_id_msb, uint8_t message_id_lsb,
                            uint8_t* data, size_t data_len)
diff --git a/communication/src/messages.h b/communication/src/messages.h
index 3824262..4cac7ec 100644
--- a/communication/src/messages.h
+++ b/communication/src/messages.h
@@ -88,6 +88,8 @@ public:
 
        static size_t event(uint8_t buf[], uint16_t message_id, const char *event_name,
                     const char *data, int ttl, EventType::Enum event_type, bool confirmable);
+       static size_t event(uint8_t buf[], uint16_t message_id, const char *event_name,
+               const char *data, size_t data_length, int ttl, EventType::Enum event_type, bool confirmable);
 
 
     static inline size_t empty_ack(unsigned char *buf,
diff --git a/communication/src/protocol.h b/communication/src/protocol.h
index d4c19b5..2bda7be 100644
--- a/communication/src/protocol.h
+++ b/communication/src/protocol.h
@@ -367,6 +367,24 @@ public:
[... 188 lines cut ...]
 #endif
 }
+
+Future<bool> CloudClass::publish_event(const char *eventName, const char *eventData, size_t data_length, int ttl, PublishFlags flags) {
+#ifndef SPARK_NO_CLOUD
+    spark_send_event_data d = { sizeof(spark_send_event_data) };
+
+    // Completion handler
+    Promise<bool> p;
+    d.handler_callback = publishCompletionCallback;
+    d.handler_data = p.dataPtr();
+
+    if (!spark_send_bin_event(eventName, eventData, data_length, ttl, flags.value(), &d) && !p.isDone()) {
+        // Set generic error code in case completion callback wasn't invoked for some reason
+        p.setError(Error::UNKNOWN);
+        p.fromDataPtr(d.handler_data); // Free wrapper object
+    }
+
+    return p.future();
+#else
+    return Future<bool>(Error::NOT_SUPPORTED);
+#endif
+}

```

m-mcgowan, 2018-06-05:
Thanks for the contribution!  Rather than pasting diffs here, could you please submit a Pull Request to this github repo. That will make it easier for us to review.  Also, a test plan would be useful so we can validate the expected behaviour.

kubark42, 2019-04-27:
@m-mcgowan, it looks like @aeris-ming might have moved on, but this feature is still pretty important. It does seem very critical to optimized IoT tasks to be able to publish data in the binary format of one's choosing. With the mesh functionality which depends crucially on the `Particle.publish()` call-- it has no other reliable and code agnostic way to get data to the cloud-- binary TX is a necessity. High availability networks which include a cellular node will have a strong financial motivator to send the minimum amount of data necessary.

The upshot is that it would be great if this feature could work its way to full implementation.

laughlinbarker, 2020-02-26:
I'm considering the ParticleIO for a system that currently utilizes Iridium SBDs.  Given this is an open issue, is it accurate to assume there is still no means to send arbitrary binary data up to the 255 byte limit?

This strikes me as a pretty significant drawback. Can you comment on weather this feature can be expected anytime soon?

Best regards,