particle-iot/device-os #1494: button_status event does not fire in Listening Mode

kdhooper, 2018-03-05:
## Bug Report

### Expected Behavior

`button_status` event should be called while in Listening Mode on Photon/P0/P1 when `SYSTEM_THREAD(ENABLED);` is used and the Setup button is pressed and released.

### Observed Behavior

Observed on P1 hardware.

While some other event callbacks seem to work during Listening Mode, (particularly `setup_update`), `button_status` seems to be called on release when manually entering Listening Mode and never called again regardless of whether it's being pushed.

### Steps to Reproduce

Use the given test code and observe the `button_status` stop working when manually entering Listening Mode.

(NOTE: There is another bug that suppresses Logger logging within `button_status`, so just use `Serial.println()`.)

### Test App

```c
#include <Particle.h>

SYSTEM_THREAD(ENABLED);

void eventSetupModeActive() {Serial.println("eventSetupModeActive()");}

void eventSetupButtonPressOrRelease(system_event_t event, int pressDuration) {
  Serial.println("eventSetupButtonPressOrRelease()");
}

void setup() {
  Serial.begin(115200);
  System.on(setup_begin+setup_update, eventSetupModeActive);
  System.on(button_status, eventSetupButtonPressOrRelease);
}

void loop() {

}
```

### References

https://community.particle.io/t/setup-button-status-event-doesnt-fire-while-wifi-listen-is-active/40033
https://community.particle.io/t/system-on-events-in-listening-mode/39130

sergeuz, 2018-03-05:
This is probably due to the fact that some button events are generated in the context of an ISR, which breaks detection of the current thread in the eventing system: https://github.com/particle-iot/firmware/issues/1148.

We now have a pooled memory allocator, so it should be pretty straightforward to implement safe queuing for button and other events generated in ISR. cc @m-mcgowan

technobly, 2019-01-11:
FYI: This is still an issue despite the changes made in 0.8.0-rc.12 for button handling.