particle-iot/device-os #1499: Please changed the behavior of Serial.write()

aeris-ming, 2018-03-09:
Now Serial.wrtie() will block or over overwritten.
It will be very nice if Serial.write() write as possible as it could and then return the number of bytes written without any block.

m-mcgowan, 2018-03-09:
Have you seen `availableForWrite()`? It tells you how many characters can be written without blocking. https://docs.particle.io/reference/firmware/photon/#availableforwrite-

dmiddlecamp, 2018-03-09:
Isn't this behavior also configurable? -> https://docs.particle.io/reference/firmware/electron/#blockonoverrun-

aeris-ming, 2018-03-20:
@m-mcgowan I think it is not a bad idea if Serial.write() can support it,So user do not have to write their own function and I think it will be easy to handle race condition too.
static int32_t uart1_write(uint8_t *buffer, int32_t size, int32_t id)
{
        int32_t aw = Serial1.availableForWrite();
        aw = (size >= aw) ? aw : size;

        return Serial1.write(buffer, aw);
}

aeris-ming, 2018-03-27:
it seems no more people support it

closed 2018-03-27 by aeris-ming