particle-iot/device-os #1502: How to use nanopb in user code

elcojacobs, 2018-03-17:
<details>
  <summary><i>I was using nanopb in my user code, since updating to the photon-tcp-fixes branch based on recent develop, I have make errors.</i></summary>
</details>

Using my own copy of nanopb in a directory of my choosing and adding the c files to CSRC doesn't work. It clashes with the files in the particlesystem directory.
So I removed nanopb from my app dependencies and pointed my makefiles to the folder in the particle system firmware.

In my user makefile, I have:
```
# add nanopb dependencies
include $(SOURCE_PATH)/platform/spark/firmware/nanopb/import.mk
```

I am getting undefined references to `pb_decode_delimited_noinit` and `pb_encode_delimited`.
Some functions from pb_encode.c and pb_decode.c are exported in `services2_dynalib.h`, but not these two.

How do I fix this issue?

aeris-ming, 2018-03-19:
I am trying to use nanopd too.
nanopd is using binary data,But parameter of Particle.function() is String.
Are you sure It is OK?

elcojacobs, 2018-03-19:
I think that's out of the scope of this issue and should be asked on the forum.

To resolve the issue with some nanopb functions being available at link time, but not all, is possible to add all nanopb functions that are not part of dynalib to separate library that can be included at link time? @m-mcgowan @avtolstoy

elcojacobs, 2018-03-20:
I have duplicated necessary functions to a user space c-file as a workaround:

https://github.com/BrewPi/firmware/commit/d410b6cc23a35fa5d7a2864ab69b854b9a53a8c9

If you think this is something that end users should have to do, feel free to close this issue.
If you think this should be provided by the particle framework, leave it open.

aeris-ming, 2018-03-21:
nanopb it is an application protocol if system-lib not depend on it I think It should maintain by users

bkobkobko, 2018-03-21:
Hi @aeris-ming 

Your question about Particle function has nothing to do with this github issue.  That is not a best practice.

You got the full and correct answer in the forum--you must encode binary data today.  There is an issue to request binary data here on github that you can add your voice to the request, but today there is no way to send binary data except by encoding it.

aeris-ming, 2018-03-21:
@bkobkobko Thanks!

m-mcgowan, 2018-05-02:
The application module has a dependency on `services`, which is where the nanopb dynalib code resides.
In hindsight, I think the nanopb should have been it's own dynalib module so that it could be specifically included where needed.

Going forward, we can make it an optional import and include it only if explicitly requested (E.g. PARTICLE_NANOPB is defined). This would work by changing the dynalib import to not create stubs for the nanopb functions (but still increment the offset counter).  I feel the same should be done for all our 3rd party libraries exposed through dynalib (nanopb, mbedtls, jsmn).

We may need to do something with headers too to prevent version clashes and such with application headers.

elcojacobs, 2018-05-03:
It's actually great to be able to use the nanopb dependencies are already in the system layer, this saves the application code space.
The only issue is that the header files define some functions that are linked into the system layer and others that are not. This creates an issue for the user if he wants to use the functions that are not linked (undefined reference).
If Particle would split the compiled sources into a dynamically linked part that particle needs (in system) and a library that is linked at compile for the stuff that particle doesn't need, this would solve the issue. The statically linked library will be linked into the user application space. This would work let the linker solve the issue without user action required.

m-mcgowan, 2018-05-03:
Perhaps adding the missing functions as a Particle Library would be the simplest route forward. In the back of my mind I am thinking about issues with version mismatches between the library and system firmware, but that can be dealt with since each system firmware provides version defines.

JonKimbel, 2020-01-19:
I just ran into this and duplicated elcojacob's workaround, I'm a bit worried it'll break at some point in the future when my Photon's firmware updates - perhaps this isn't something to worry about though, I don't really know how the device-os/user code split works.

Any thoughts on this bug? Do I need to worry about firmware updates killing the workaround?

avtolstoy, 2020-01-21:
I don't think we can say with 100% of certainty that things won't break in the later DeviceOS releases. We are exploring a couple of options on how to deal with 3rdparty libraries in user code and as discussed above in this issue, one of the solutions might be that we'll hide them by default.