particle-iot/device-os #1506: Enhancement for LogHandler: Allow incremental logging

ScruffR, 2018-03-23:
Especially to produce verbose trace logs it would be nice *not* to create a new line for every call of `Log.trace()` but rather build up the output in an intermediate buffer and then flush that to the LogHandler in one go.
Sure it would be possible to build the log string in a special variable and then only commit the "final" string to the LogHandler but that would require flash and RAM even if logging is not enabled.

If we had something like 
`Log.traceDeferred(...)` and `Log.flush()` something like this would *not* produce multiple lines of log entries
```
  for (i = 0; i < bytes; i++) {
    while (!cameraStream.available()) yield(); // wait for bytes
    s = cameraStream.read();
    Log.trace("%02x ", s);
    // Skip first 4 and last 2, Page 10 of the datasheet
    if (i >= 4 && i < bytes - 2) {
      this->imgBuffer[i - 4] = s;
      this->image_pos--;
    }
  }
```

sergeuz, 2018-03-23:
Hey @ScruffR, thanks for the feature request. It is unlikely that we will implement this feature, since we want to keep the API simple and avoid dynamic memory allocations in the logging service. Also, the proposed extension to the API doesn't seem to be thread-safe. As a workaround you can try so-called direct logging, which kind of does what you want already:
```cpp
for (i = 0; i < bytes; i++) {
    while (!cameraStream.available()) yield(); // wait for bytes
    s = cameraStream.read();
    Log.printf("%02x ", s);
    // Skip first 4 and last 2, Page 10 of the datasheet
    if (i >= 4 && i < bytes - 2) {
        this->imgBuffer[i - 4] = s;
        this->image_pos--;
    }
}
Log.print("\r\n"); // Add new line to the log
```

technobly, 2018-03-23:
FYI you can also specify log level with these direct logging functions: https://github.com/particle-iot/firmware/blob/develop/wiring/inc/spark_wiring_logging.h#L293-L319

```cpp
Log.printf(LOG_LEVEL_INFO, "%02x ", s);
Log.print(LOG_LEVEL_INFO, "\r\n");
```

ScruffR, 2018-03-24:
Didn't know about that :blush: 
Good to know - for my use-case this is enought and this issue can be closed (unless you want it open for the `documentation` tag).

closed 2020-03-03 by avtolstoy