particle-iot/device-os #1513: 0.7.0 Application Watchdog causes Assertion Failure panic crash

ScruffR, 2018-04-01:
As outlined in this thread
https://community.particle.io/t/application-watchdog-test-results-in-assertion-failure/40765?u=scruffr

This simple sketch (pretty close to the docs sample) does cause a SOS+10 panic.

```cpp
SYSTEM_MODE(SEMI_AUTOMATIC)
//SYSTEM_THREAD(ENABLED)    // optional but doesn't change a thing

ApplicationWatchdog wd(10000, System.reset);
void setup() { }
void loop() 
{
    int x;
    while (1) 
    {
        x++;
    }
}
```

jboalml, 2018-05-14:
In case someone is low on memory as I am, with 0.8.0-rc.4 it works with just 844 bytes of stack. Assertion failure still occurs with 843 bytes or less.

sergeuz, 2018-12-14:
Thanks for reporting this issue! It should be fixed in `develop` now: https://github.com/particle-iot/firmware/pull/1640

closed 2018-12-14 by sergeuz