particle-iot/device-os #1523: attachInterruptDirect() doesn't work as expected

ScruffR, 2018-04-11:
As it seems the newly added feature `attachInterruptDirect()` does not work as expected (or at all :wink: )

When running this test code
```
const bool direct  = true;
const int  pinTrig = A1;        
const int  pinIRQ  = D1;            // also tested  D4
IRQn_Type  exti    = EXTI9_5_IRQn;  //              EXTI3_IRQn      

void ISR() {
  pinResetFast(pinTrig); 
  pinSetFast(D7);                                           // show when triggered at least once
}

void setup() {
  pinMode(D7, OUTPUT);
  pinMode(pinTrig, OUTPUT);    
  pinMode(pinIRQ , INPUT);     
  if (direct)
    attachInterruptDirect(exti, ISR);
  else
    attachInterrupt(pinIRQ, ISR, RISING);
}

void loop() {
  static uint32_t msTimeout = millis();

  if (digitalRead(pinTrig) && millis() - msTimeout < 10) {  // still high after 10ms
    msTimeout = millis();
    digitalWrite(pinTrig, LOW);                             // reset the trigger
    delay(10);                                              // for 10ms
  } 

  digitalWrite(pinTrig, HIGH);
}
```
with `direct = false` the legacy `attachInterrupt()` approach makes the D7 LED come on and I get permanent pulses on the `pinTrig` which is triggered from `loop()` and reset in the `ISR()`. 
But with `direct = true` `attachInterruptDirect()` is used and nothing happens on D7 nor the expected pulsing on `pinTrig`.

Surprisingly `D1`/`EXTI9_5_IRQn` is also used in the test rig for this feature, but while the test must have passed, my simple test doesn't.

ScruffR, 2018-05-20:
No response yet?

pkourany, 2018-06-20:
I ran a modified version of SparkIntervalTimer which worked just fine.  I just didn't test the latency timing yet.

ScruffR, 2018-06-21:
Maybe timer interrupts are treated better than pin interrupts.
Can you check pin interrupts on one of your devices?
Or do you see a flaw in my test code?

pkourany, 2018-06-21:
@ScruffR, your ISR code should be clearing the interrupt flag as in the test code using:
```
	static const uint16_t exti_line = pinmap[pin].gpio_pin;
        ...
	// attach a direct handler
	bool res = attachInterruptDirect(irqn, []() {
		attachInterruptDirectHandler = true;
		**EXTI_ClearFlag(exti_line);**
	});
```

ScruffR, 2018-06-21:
Thanks Paul for the input.
I'll give it a try.

That's why we'd need a nice docu for new features - the prerelease docs page we used to have seems gone (still).

technobly, 2018-06-21:
This one? https://prerelease-docs.particle.io/reference/firmware/electron/#attachinteruptdirect-

avtolstoy, 2020-03-03:
This is a documentation issue.

closed 2020-03-03 by avtolstoy

ScruffR, 2020-03-03:
Was this carried forward to the docu repo? (Link?)

avtolstoy, 2020-03-03:
There is an internal issue for it.