particle-iot/device-os #1525: detachInterrupt() causing system freeze

ScruffR, 2018-04-14:
After some extensive testing (see [thread](https://community.particle.io/t/electron-freezes-with-solid-green-or-cyan-led/41022/22
)) there seems to be a problem with `detachInterrupt()` that's causing the system to hang/deadlock.

This test code illustrates the problem after some unpredictable but finite time (30sec ~ 15min usually).
Tested on Electron 0.6.1, 0.6.4 & 0.7.0
```
//SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);
STARTUP(System.enableFeature(FEATURE_RETAINED_MEMORY));

ApplicationWatchdog wd(10000, System.reset, 1536);

enum SENSOR_STATUS 
{ SENSOR_FAIL                   = 2
, SENSOR_SUCCESS                = 3
};

const uint32_t    msMeasuring   =  3000;
const uint32_t    msRefresh     =     0; // as rapid as possible 10000;
const int         pinData       = D3;
const int         pinPower      = C0;
const int         pinPWM        = D0;           // for isolated test bridge pinPWM to pinData

volatile bool     doCount       = false;
volatile uint32_t rpmCount;
uint32_t          rpm;
uint32_t          timerStart;
uint32_t          timerStop;
SENSOR_STATUS     status;

retained int      lastLocation  = 0;
retained uint32_t lastTime      = 0;
int               lastCrash     = 0;
uint32_t          lastCrashTime = 0;
void getRpm();
void interrupt();

void setup() {
  lastCrash     = lastLocation;
  lastCrashTime = lastTime;
  
  Serial.begin();
  pinMode(pinData , INPUT);
  pinMode(pinPower, OUTPUT);
  pinMode(pinPWM  , OUTPUT);                    // for isolated test
}

void loop() {
  static uint32_t msLastRefresh = 0;
  static uint32_t pwmFrequency  = 0;
  if (millis() - msLastRefresh >= msRefresh) {
    getRpm();
    Serial.printlnf("RPM: %5lu, PWM: %5lu, status: %s, millis: %10lu, lastCrash: %d (%10lu)", 
                    rpm, pwmFrequency, status == SENSOR_SUCCESS ? "OK" : "XX", millis(), lastCrash, lastCrashTime);
    pwmFrequency = random(1, 65535);
    analogWrite(pinPWM, 1, pwmFrequency);       // set the test frequency for next turn
    Serial.printf("next PWM: %5lu\t-> ", pwmFrequency);
    Serial.flush();
    msLastRefresh = millis();
  }
}

void getRpm() {                                                 //  checkpoins to locate the offending line 
                                                                    lastLocation = 0;
  pinResetFast(pinPower);
  delay(50);
  rpmCount   = 0;
  timerStart = millis();
  doCount = true;
                                                                    lastLocation++; lastTime = millis();
  attachInterrupt(pinData, interrupt, RISING);
                                                                    lastLocation++; lastTime = millis();
  while(millis() - timerStart < msMeasuring) Particle.process();
                                                                    lastLocation++; lastTime = millis();
  timerStop = millis();
                                                                    lastLocation++; lastTime = millis();
  noInterrupts();
                                                                    lastLocation++; lastTime = millis();
  detachInterrupt(pinData);
                                                                    lastLocation++; lastTime = millis();
  interrupts();                                                 // offending line    
                                                                    lastLocation++; lastTime = millis();
  rpm    = (1000 * rpmCount) / (timerStop - timerStart);
  status = (0 <= rpm && rpm < 5000) ? SENSOR_SUCCESS : SENSOR_FAIL;
  pinSetFast(pinPower);
}

void interrupt() {
  rpmCount++;
}
```
After a freeze, the retained variables show which line of code was executed last and will be printed via USB Serial.
With the code as is the last checkpoint reached is the one _before_ `interrupts()`
But when not using `noInterrupts()`/`interrupts()` the last checkpoint reached is _before_ `detachInterrupt()`.
Also when wrapping `detachInterrupt()` into an `ATOMIC_BLOCK` the last checkpoint reached is _before_ that `ATOMIC_BLOCK`.

Not surprisingly the Application Watchdog doesn't fire either, hence [HW watchdog](https://github.com/particle-iot/firmware/issues/1382) is needed.

ScruffR, 2018-07-03:
Bump for the 0.7.0 issue resove sprint.

pkourany, 2018-08-08:
@ScruffR, did you try the code WITHOUT the `noInterrupts()/interrupts()`?

ScruffR, 2018-08-08:
Of course: 

> But when not using noInterrupts()/interrupts() the last checkpoint reached is before detachInterrupt().

hwestbrook, 2019-03-28:
This isn't an issue with firmware post 0.7.0?

ScruffR, 2019-03-28:
I haven't tested since, but 0.7.0 was the most recent version back then and since there was no official response to this issue report I'm not sure whether it was ever addressed.

hwestbrook, 2019-03-28:
Maybe you should remove the 0.7.0 heading? Seems misleading