particle-iot/device-os #1526: 0.6.3 to 0.7.0 goes wrong with at least 5 out of 5 devices I own

kasperkamperman, 2018-04-14:
## Bug Report
I hope this issue can be fixed. I'm getting a bit disappointed in the support of Particle. First it took really long before 0.7.0 was released and now it fails on all my devices. In the communication you state that the Photon will be still supported. I hope you do and that not al the resources go now to the 3rd generation of devices. 

I gave detailed debug steps in this forum post. After that it stays quiet. I hope that submitting this issue will help to get this problem under the attention again:
https://community.particle.io/t/update-to-0-7-0-goes-wrong/

### Expected Behaviour

Connect to the Particle cloud. 

### Observed Behavior

Don't connect to the Particle cloud. 

### Steps to Reproduce

Online
- Uploading a file with the Particle Cloud (build.particle.io) to a 0.6.3 targeted at 0.7.0.
- It auto updates the firmware. When it's done, it will never connect to the cloud again. 

Offline:
- Same story trying to fix the device with Particle doctor. 

### Test App

No applicable

### References

Links to the Community, Docs, Other Issues, etc..

kasperkamperman, 2018-04-15:
Hi ScruffR,
I think I described and did what you suggest. But I made it more clear, what I said a start from 0.6.3 Tinker (changed my description above to make that more clear). 

I'll try downgrading in a different order, however that's not the main problem. The main problem is that update to 0.7.0 fails. I don't prefer to downgrade at all.

ScruffR, 2018-04-15:
PMed for further investigation of possible cause.

kasperkamperman, 2018-05-19:
Trials as communicated with ScruffR. I hope some progress will be shared by Particle, because until now it stays pretty silent. In way like it works in our testing networks, so the problem is kind of non-existant. 

## Mobile hotspot tryout 

Ok, I could get a device to update with my mobile hotspot. 

Indeed I got the 5 safe-mode-updater events. I'll need to check at home in which step I get stuck. 

```
event: spark/safe-mode-updater/updating
data: {"data":"1","ttl":60,"published_at":"2018-04-16T12:27:27.761Z","coreid":"particle-internal"}

event: spark/safe-mode-updater/updating
data: {"data":"2","ttl":60,"published_at":"2018-04-16T12:28:03.768Z","coreid":"particle-internal"}

event: spark/safe-mode-updater/updating
data: {"data":"0","ttl":60,"published_at":"2018-04-16T12:28:44.590Z","coreid":"particle-internal"}

event: spark/safe-mode-updater/updating
data: {"data":"1","ttl":60,"published_at":"2018-04-16T12:29:20.400Z","coreid":"particle-internal"}

event: spark/safe-mode-updater/updating
data: {"data":"2","ttl":60,"published_at":"2018-04-16T12:29:54.196Z","coreid":"particle-internal"}
```

Ok, I found some time to test it again on my home network. It does only the first two steps, then it hangs...
```
event: spark/safe-mode-updater/updating
data: {"data":"1","ttl":60,"published_at":"2018-04-18T21:44:48.004Z","coreid":"particle-internal"}

event: spark/safe-mode-updater/updating
data: {"data":"2","ttl":60,"published_at":"2018-04-18T21:45:24.352Z","coreid":"particle-internal"}
```
If I then power on my mobile phone hotspot, it continues. 

It's interesting to see what's going on. I can imagine more Particle devices being used in home networks. My network is stable and I'm happily surfing on the net with multiple devices on high speed. 

Your DHCP theory might work, however if I revert to 0.6.3 it directly connects to the cloud. 

How can I help to test? Might it be worth it to set a static Ip in the device (https://docs.particle.io/reference/firmware/photon/#setstaticip-). I did this with my WiFi credentials now as well, I think it's stored in the EEPROM so even with firmware updates the IP should stay static I think?

Would that be a good test to see if it's something related to the DHCP?

## Try-out by setting a static ip like below messed up my Photon (can't get it online even not when going back to 0.6.3)

I've build the following code to set my device to a static ip based on the ip information from the DHCP:
https://go.particle.io/shared_apps/5ae83294105a0839020011fe

My idea is to see what happened with a static ip configured. 

Running the script gives me the following result:
my current IP address:  192.168.1.18
my current subnet Mask: 255.255.255.0
my current gateway IP:  192.168.1.1
my current dns IP:      0.0.0.0

After I did this, I did another reboot to see if the static ip would be used. This fails. The device won't connect to the cloud anymore. It blinks rapidly light blue, with 3 orange blinks in between. 

I tried updating the Photon locally to use dynamicIP with the code below, however that won't work either. 
- Replacing the system firmware not. 
- Particle doctor also doesn't work after setting antenna to internal I get:
```? Select the antenna to use to connect to Wi-Fi Internal
The Doctor didn't complete sucesfully. undefined
```
- I tried skipping the steps untill set IP, this gives the following error:
```
? Select how the device will be assigned an IP address Dynamic IP
The Doctor didn't complete sucesfully. "path" is not defined: undefined
> Please visit our community forums for help with this error:
```

Any other way to reset the WiFi chip to use dynamic IP?

```
#include "Particle.h"

IPAddress myAddress;
IPAddress netmask;
IPAddress gateway;
IPAddress dns;

String firmwareVersion;
String publishInformation;

void setup() {

    //while(!Serial.available())
    //Particle.process();

    WiFi.useDynamicIP();

    myAddress = WiFi.localIP();
    netmask   = WiFi.subnetMask();
    gateway   = WiFi.gatewayIP();
    dns       = WiFi.dnsServerIP();

    firmwareVersion = System.version().c_str();

    publishInformation += "my current firmware:    " + firmwareVersion + "\n";
    publishInformation  = "my current IP address:  " + String(myAddress) + "\n";
    publishInformation += "my current subnet Mask: " + String(netmask) + "\n";
    publishInformation += "my current gateway IP:  " + String(gateway) + "\n";
    publishInformation += "my current dns IP:      " + String(dns) + "\n";

    //WiFi.setStaticIP(myAddress, netmask, gateway, dns);

    // now let's use the configured IP
    //WiFi.useDynamicIP();
    //Serial.println("Static IP is set");

    Particle.publish("address info", publishInformation, 60, PRIVATE);
    Particle.variable("getInfo", publishInformation);

    Serial.begin(9600);
}

void loop() {
    digitalWrite(D7, HIGH);

    Serial.println(publishInformation);

    // Prints out the local IP over Serial.
    //Serial.println("my current IP address:  " + myAddress);
    //Serial.println("my current subnet Mask: " + netmask);
    //Serial.println("my current gateway IP:  " + gateway);
    //Serial.println("my current dns IP:      " + dns);

    delay(1000);
    digitalWrite(D7, LOW);
    delay(1000);
}

```

kasperkamperman, 2018-05-21:
The device number I have still trouble with is 410024000447343339373536. This gives me a lightblue blink and three orange flashes. What does this mean? 

I think the static ip update before updating to new firmware script is a bad idea or not? Or is there a proper way to do it?

m-mcgowan, 2018-05-21:
You should not need to set a static IP, and I feel that will only compound the issues in general, although if there was a specific reason why you think that might have been necessary in this case, I'd love to hear it. 

The 3 orange flashes indicates a problem where the device can connect to the cloud but not complete a successful handshake. The instructions here show you how to fix this. https://community.particle.io/t/photon-flashing-cyan-and-blinking-orange/21530/6?u=mdma. 

Before following these instructions, it's a good idea to take a copy of the DCT region for later analysis - 
```
dfu-util -d 2b04:d006 -a 0 -s 0x8004000:0x8000 -U 3_orange_flashes.bin
```

I hope that helps!

ScruffR, 2019-02-15:
@m-mcgowan, a recent post may have revealed some link between this issue and #1556

https://community.particle.io/t/update-to-0-7-0-goes-wrong/40700/56?u=scruffr

kasperkamperman, 2019-02-18:
I can confirm. The mode of my router is n only:

> 802.11 Mode:802.11n only

technobly, 2019-04-26:
@kasperkamperman if you have the ability to check #1755 to see if it resolves the issue for you, your feedback would be appreciated.

PR #1755 (2019-04-29): [Photon/P1] Fixes 802.11n mode
### Problem

#1556

### Solution

particle-iot/photon-wiced#32

The PR above resolves and incompatibility in WICED code when handling country codes with the current (newer, with KRACK2 security bugfixes) version of CYW43362 firmware we've introduced in #1420, #1419, #1418.

### Steps to Test

Attempt to connect to 802.11n-only AP. The connection should succeed.

### Example App

N/A

### References

- Closes #1526 
- Closes #1556 
- particle-iot/photon-wiced#32
- [CH17929]

---

### Completeness

- [x] User is totes amazing for contributing!
- [x] Contributor has [... cut]
closed 2019-04-29 by avtolstoy