particle-iot/device-os #1546: 0.7.0 ParticleSoftSerial causes System.sleep(pin,....) to prematurely wake

rvnash, 2018-06-04:
On the Particle Photon, the function System.sleep(pin, RISING, 60); does not reliable sleep when the ParticleSoftSerial is being used.

In this code the the call to System.sleep usually will abort the sleep nearly immediately.

```cpp
#include <Particle.h>
SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);
#include <ParticleSoftSerial.h>

uint32_t lastSleep;
#define SLEEP_PIN D1

ParticleSoftSerial ss(D4,D5);

void setup()
{
  WiFi.off();
  Serial.begin();
  ss.begin(9600); // Comment out this line and sleep/wakeup works
  pinMode(SLEEP_PIN,INPUT_PULLDOWN);
  lastSleep = millis();
}

void loop()
{
  if (millis()-lastSleep > 10000) {
    Serial.println("Going to sleep");
    delay(1000);
    System.sleep(SLEEP_PIN,RISING,10);
    delay(1000);
    Serial.println("Waking up");
    lastSleep = millis();
  }
}
```

Changing the System.sleep line to these 3 lines:
```cpp
    ss.end();
    System.sleep(SLEEP_PIN,RISING,10);
    ss.begin(9600);
```

causes the sleep to behave as expected.

I suspect that attachInterrupt to the Rx pin inside ParticleSoftSerial's begin method is what causes this behavior. Perhaps this example could be simplified to demonstrate that only the attachInterrupt is the culprit. However, I did not run that experiment.

ScruffR, 2018-06-05:
I'd think `ParticleSoftSerial` is not the actual cause but just one possible "victim" to a potentially more general issue of pin interrupts (other than the designated wake pin) interfering with the proper functioning of `System.sleep()`.

`ParticleSoftSerial` uses `attachInterrupt()` on the designated RX pin and once a start bit is received or some bytes to transfer are present in the TX buffer a `SparkIntervalTimer` is started to create the timing for the transfer. 

So either `attachInterrupt()` or the HW timer interrupts used via `SparkIntervalTimer` might upset `System.sleep()` but I'd think `System.sleep()` should be more resilient or its limitations should be clearly documented.

However, since user code knows when it intends to sleep, calling `ParticleSoftSerial::end()` would be good practice anyway.

pkourany, 2018-08-08:
@ScruffR, any enabled EXTI interrupt will wake the device in this sleep mode.  So you could add a `disable()/enable()` set of functions in ParticleSoftSerial to be called before sleeping.  However, calling `end()/begin()` is better practice as you point out.

ScruffR, 2018-08-08:
> any enabled EXTI interrupt will wake the device in this sleep mode.

I'm not sure about that, but I would need to test.
This assumption was once brought up by Mat but subsequent tests did render them not true. 
Things may have chenged tho'