particle-iot/device-os #1548: Electron power-on connectivity issues with 0.8.0-rc

kubark42, 2018-06-13:
## Bug Report

### Expected Behavior

Stay connected without an external battery

### Observed Behavior

Upgrading to 0.8.0-rc7 has made it so our units no longer stay connected when booted off the power supply. We are using a regulated lab 5V supply rated well in excess of 5A.

Another user reported similar issues with 0.8.0-rc3.

This was reproduced across multiple Electrons.

### Steps to Reproduce

The Electrons boot and stay connected when a battery is present; but without the battery, on the first boot they do not stay connected. What’s weird is that they do complete the handshake and deliver an initial diagnostics message (we can see it in the console), but then they go silent and become unreachable. The cyan LED starts blinking rapidly.

It’s also weird that the battery can be removed and then the Electron rebooted without reproducing this problem. It really is only on initial power-on.

Using particle device doctor to downgrade to 0.7.0 solves the problem.

This occurs whether the firmware was compiled for 0.7.0 or 0.8.0-rc4 (-rc7 is not available right now as an option in Particle Dev).

This is using a 3rd party AT&T SIM, with KeepAlive set at 60 seconds.

### Test App

None yet.

### References

https://community.particle.io/t/electron-power-on-connectivity-issues-with-0-8-0-rc/42391/

technobly, 2018-07-03:
I have no doubt this is a firmware issue of some sort that we will look into.  

I just wanted to mention that the Electron is not designed to be operated without the battery unless you have provided extra attention to the power supply requirements (10W of power and bulk capacitance added at VIN to handle the peak currents required).

https://docs.particle.io/datasheets/electron-(cellular)/e-series-datasheet/#vin
>The input voltage range on VIN pin is 3.88VDC to 12VDC. When powering from the VIN pin alone, make sure that the power supply is rated at 10W (for example 5 VDC at 2 Amp). If the power source is unable to meet this requirement, you'll need connect the LiPo battery as well. An additional bulk capacitance of 470uF to 1000uF should be added to the VIN input when the LiPo Battery is disconnected. The amount of capacitance required will depend on the ability of the power supply to deliver peak currents to the cellular modem.

kubark42, 2018-09-05:
> I just wanted to mention that the Electron is not designed to be operated without the battery unless you have provided extra attention to the power supply requirements (10W of power and bulk capacitance added at VIN to handle the peak currents required).

Understood. As mentioned in the above, these guidelines were followed with all power supplies. Our power supply is capable of 25W, and maintaining voltage when undergoing fast transients.

PR #1581 (2018-12-05): [Electron/LTE] Power Manager Watchdog Timer Fix
### Problem

Modem on Electron/LTE turn off unexpectedly during initialization when powered from a USB Host only (no battery).

After an investigation it was discovered that the PMIC watchdog causes current limit to drop to 100mA temporarily which causes the modem to stop responding to AT commands.

### Solution

Power Manager no longer using PMIC watchdog because it causes current limit to drop to 100mA temporarily when it expires

**Workaround:** For 0.8.0-rc.1 ~ 0.8.0-rc.11, add `PMIC().disableCharging()` to `setup()` when no battery is used.
 [... cut]
closed 2018-12-05 by sergeuz