particle-iot/device-os #1551: errors when make all PLATFORM=gcc

beriberikix, 2018-06-20:
I'm to build a sample using `PLATFORM=gcc` but are running into errors. Using the develop branch of `firmware,` I tried the following:
```
 $ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.3.1 20160307 (release) [ARM/embedded-5-branch revision 234589]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$ pwd
firmware/modules
$ make all PLATFORM=gcc
makefile:20: *** "Platform 'gcc' does not support dynamic modules".  Stop.
```
and
```
$ pwd
firmware
$ make all PLATFORM=gcc
Building firmware for GCC xcompile, platform ID: 3, product ID: 3
...
Building c file: src/rgbled.c
Invoking: ARM GCC C Compiler
mkdir -p ../build/target/services/platform-3/./src/
gcc -DPLATFORM_THREADING=0 -DPLATFORM_ID=3 -DPLATFORM_NAME=gcc -DUSBD_VID_SPARK=0x1D50 -DUSBD_PID_DFU=0x607F -DUSBD_PID_CDC=0x607D -g3 -m64 -O3 -gdwarf-2 -Wno-unused-local-typedefs -DSYSTEM_VERSION_STRING=0.7.0 -DRELEASE_BUILD -Werror -I./inc -I../hal/inc -I../hal/shared -I../wiring/inc -I../dynalib/inc -I. -MD -MP -MF ../build/target/services/platform-3/./src/rgbled.o.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -D_WINSOCK_H -D_GNU_SOURCE -DLOG_MODULE_CATEGORY="\"service\"" -Wno-pointer-sign -std=gnu99 -c -o ../build/target/services/platform-3/./src/rgbled.o src/rgbled.c
src/rgbled.c:19:5: error: 'MODULE_FUNCTION' is not defined, evaluates to 0 [-Werror,-Wundef]
#if MODULE_FUNCTION != MOD_FUNC_BOOTLOADER
    ^
1 error generated.
make[1]: *** [../build/target/services/platform-3/./src/rgbled.o] Error 1
make: *** [services] Error 2
```
and
```
$ pwd
firmware/main
$ make all PLATFORM=gcc
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C ../newlib_nano 
make[1]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C ../user 
make[1]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C ../wiring 
Building cpp file: src/spark_wiring.cpp
Invoking: ARM GCC CPP Compiler
mkdir -p ../build/target/wiring/platform-3/src/
gcc -DPLATFORM_THREADING=0 -DPLATFORM_ID=3 -DPLATFORM_NAME=gcc -DUSBD_VID_SPARK=0x1D50 -DUSBD_PID_DFU=0x607F -DUSBD_PID_CDC=0x607D -g3 -m64 -O3 -gdwarf-2 -Wno-unused-local-typedefs -DPRODUCT_ID=3 -DPRODUCT_FIRMWARE_VERSION=65535 -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -DSYSTEM_VERSION_STRING=0.7.0 -DRELEASE_BUILD -Werror -I./inc -I../hal/inc -I../hal/shared -I../hal/src/gcc -I../services/inc -I../communication/src -I../system/inc -I../platform/shared/inc -I../platform/MCU/gcc/inc -I../dynalib/inc -I. -MD -MP -MF ../build/target/wiring/platform-3/src/spark_wiring.o.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -fno-builtin-malloc -fno-builtin-free -fno-builtin-realloc -DMODULE_VERSION=207 -DMODULE_FUNCTION=3 -DMODULE_DEPENDENCY=0,0,0 -DMODULE_DEPENDENCY2=0,0,0 -D_WINSOCK_H -D_GNU_SOURCE -DLOG_MODULE_CATEGORY="\"wiring\""  -std=gnu++11 -c -o ../build/target/wiring/platform-3/src/spark_wiring.o src/spark_wiring.cpp
In file included from src/spark_wiring.cpp:1:
In file included from ./inc/spark_wiring.h:42:
./inc/spark_wiring_wifi.h:203:29: error: suggest braces around initialization of subobject [-Werror,-Wmissing-braces]
        HAL_IPAddress ip = {0};
                            ^
                            {}
1 error generated.
make[1]: *** [../build/target/wiring/platform-3/src/spark_wiring.o] Error 1
make: *** [wiring] Error 2
```

Any ideas?

rickkas7, 2018-06-20:
You need to do the build from main, not from the top level.

```
cd main
make all platform=gcc PARTICLE_DEVELOP=1
```

Also, the gcc platform (3) uses your native gcc compiler, not the gcc-arm compiler.

```
gcc --version
```

I just did a build of the current development on the Mac with the command above and it built successfully.

beriberikix, 2018-06-20:
Oooh, that worked!

> Also, the gcc platform (3) uses your native gcc compiler

duh! 🤦‍♂️

beriberikix, 2018-06-20:
Actually, getting a linking error, you? Or should I ignore?

```
Invoking: ARM GCC C++ Linker
mkdir -p ../build/target/main/platform-0-lto/
arm-none-eabi-g++ -DSTM32_DEVICE -DSTM32F10X_MD -DPLATFORM_THREADING=0 -DPLATFORM_ID=0 -DPLATFORM_NAME=core -DUSBD_VID_SPARK=0x1D50 -DUSBD_PID_DFU=0x607F -DUSBD_PID_CDC=0x607D -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb -flto -DINCLUDE_PLATFORM=1 -DPRODUCT_ID=0 -DPRODUCT_FIRMWARE_VERSION=65535 -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -DMBEDTLS_CONFIG_FILE="<mbedtls_config.h>" -DSYSTEM_VERSION_STRING=0.7.0 -DRELEASE_BUILD -Werror -I../user/inc -I../wiring/inc -I../hal/inc -I../hal/shared -I../hal/src/core -I../hal/src/stm32 -I../system/inc -I../services/inc -I../communication/src -I../platform/shared/inc -I../platform/MCU/STM32F1xx/CMSIS/Include -I../platform/MCU/STM32F1xx/CMSIS/Device/ST/Include -I../platform/MCU/STM32F1xx/SPARK_Firmware_Driver/inc -I../platform/MCU/shared/STM32/inc -I../platform/MCU/STM32F1xx/STM32_StdPeriph_Driver/inc -I../platform/MCU/STM32F1xx/STM32_USB_Device_Driver/inc -I../platform/NET/CC3000/CC3000_Host_Driver -I../crypto/inc -I../crypto/mbedtls/include -I../dynalib/inc -I. -MD -MP -MF ../build/target/main/platform-0-lto/main.elf.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -DMODULE_VERSION=207 -DMODULE_FUNCTION=3 -DMODULE_DEPENDENCY=0,0,0 -DMODULE_DEPENDENCY2=0,0,0 -D_WINSOCK_H -D_GNU_SOURCE ../build/target/main/platform-0-lto/./src/module_info.o  ../build/target/main/platform-0-lto/startup/startup_stm32f10x_md.o --output ../build/target/main/platform-0-lto/main.elf -nostartfiles -Xlinker --gc-sections -flto -Os -fuse-linker-plugin -Tlinker_stm32f10x_md_dfu.ld -L../build/arm/linker --specs=nano.specs -lc -lnosys -u _printf_float -Wl,-Map,../build/target/main/platform-0-lto/main.map  -L../build/target/user/platform-0-lto/src/ -L../build/target/wiring/platform-0-lto/ -L../build/target/system/platform-0-lto/ -L../build/target/services/platform-0-lto/ -L../build/target/communication/platform-0-lto-prod-0/ -L../build/target/hal/platform-0-lto/ -L../build/target/platform/platform-0-lto/ -L../build/target/wiring_globals/platform-0-lto/ -L../build/target/crypto/platform-0-lto/ -L../build/target/newlib_nano/platform-0-lto -L../build/arm/linker -Wl,--whole-archive -lnewlib_nano -luser -lwiring -lhal -lsystem -lservices -lcommunication -lplatform -lwiring_globals -lcrypto -Wl,--no-whole-archive
/Users/jberi/bin/gcc-arm/gcc-arm-none-eabi-5_3-2016q1/bin/../lib/gcc/arm-none-eabi/5.3.1/../../../../arm-none-eabi/bin/ld: cannot find libc.a
collect2: error: ld returned 1 exit status
make: *** [../build/target/main/platform-0-lto/main.elf] Error 1
```

closed 2020-03-03 by avtolstoy