particle-iot/device-os #1553: Makefile Problem

cbaser, 2018-06-24:
Hello Everyone,
I have got new Photon and I am really new about it.
I have downloaded GNU tools for arm and dfu-util as well
```
arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
------------------------------------------------------------------------------------------------------------------------
Found DFU: [2b04:d006] ver=0250, devnum=37, cfg=1, intf=0, path="1-2", alt=1, name="@DCT Flash   /0x00000000/01*016Kg""
Found DFU: [2b04:d006] ver=0250, devnum=37, cfg=1, intf=0, path="1-2", alt=0, name="@Internal Flash   /0x08000000/03*016Ka,01*016Kg,01*064Kg,07*128Kg""
```
I cloned the git repository in my file and and in firmware directory I checked out latest version.
Then, I put my Photon in DFU mode ( blinks yellow).
Finally,

```
pwd
/Particle Source/firmware
cd modules
/Particle Source/firmware/modules$ make PLATFORM=photon clean all program-dfu
make: Nothing to be done for 'clean'.
make: Nothing to be done for 'all'.
make: Nothing to be done for 'program-dfu'.
```

I do not know, what I am missing. Can you please help me?

rickkas7, 2018-06-25:
This happens if there is a space in any directory above firmware. Change "Particle Source" to "ParticleSource" or "Particle_Source" and it should work properly.

cbaser, 2018-06-25:
Hello,
Thank you for your help. I changed the Directory name and re-run the command (make PLATFORM=photon clean all program-dfu) but makefile failed:

```
Building c file: MCU/STM32F2xx/SPARK_Firmware_Driver/src/system_stm32f2xx.c
Invoking: ARM GCC C Compiler
mkdir -p ../build/target/platform/platform-6-m/./MCU/STM32F2xx/SPARK_Firmware_Driver/src/
arm-none-eabi-gcc -DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_THREADING=1 -DPLATFORM_ID=6 -DPLATFORM_NAME=photon -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD006 -DUSBD_PID_CDC=0xC006 -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -DSYSTEM_VERSION_STRING=0.7.0 -DRELEASE_BUILD -Werror -I./shared/inc -I./MCU/STM32F2xx/STM32_USB_Host_Driver/inc -I./MCU/STM32F2xx/STM32_USB_OTG_Driver/inc -I./MCU/STM32F2xx/STM32_StdPeriph_Driver/inc -I./MCU/STM32F2xx/STM32_USB_Device_Driver/inc -I./MCU/STM32F2xx/SPARK_Firmware_Driver/inc -I./MCU/shared/STM32/inc -I./MCU/STM32F2xx/CMSIS/Include -I./MCU/STM32F2xx/CMSIS/Device/ST/Include -I../services/inc -I../hal/inc -I../hal/shared -I../hal/src/photon -I../hal/src/stm32f2xx -I../hal/src/stm32 -I../hal/src/photon/api -I../hal/src/photon/include -I../hal/src/photon/wiced/security/BESL/host/WICED/ -I../hal/src/photon/wiced/security/BESL/include -I../hal/src/photon/wiced/security/BESL -I../hal/src/photon/wiced/security/BESL/crypto -I../hal/src/photon/wiced/WWD/include/ -I../hal/src/photon/wiced/platform/include/ -I../hal/src/photon/wiced/platform/GCC/ -I../hal/src/photon/wiced/security/BESL/supplicant/ -I../hal/src/photon/libraries/crypto -I../dynalib/inc -I. -MD -MP -MF ../build/target/platform/platform-6-m/./MCU/STM32F2xx/SPARK_Firmware_Driver/src/system_stm32f2xx.o.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -DBOOTLOADER_SDK_3_3_0_PARTICLE -DPARTICLE_DCT_COMPATIBILITY -DUSER_FIRMWARE_IMAGE_SIZE=0x20000 -DUSER_FIRMWARE_IMAGE_LOCATION=0x80A0000 -DMODULAR_FIRMWARE=1 -DCRYPTO_PART1_SIZE_OPTIMIZATIONS -DMODULE_VERSION=207 -DMODULE_FUNCTION=4 -DMODULE_INDEX=1 -DMODULE_DEPENDENCY=0,0,0 -DMODULE_DEPENDENCY2=0,0,0 -D_WINSOCK_H -D_GNU_SOURCE  -Wno-pointer-sign -std=gnu99 -c -o ../build/target/platform/platform-6-m/./MCU/STM32F2xx/SPARK_Firmware_Driver/src/system_stm32f2xx.o MCU/STM32F2xx/SPARK_Firmware_Driver/src/system_stm32f2xx.c
MCU/STM32F2xx/SPARK_Firmware_Driver/src/system_stm32f2xx.c: In function 'SetSysClock':
MCU/STM32F2xx/SPARK_Firmware_Driver/src/system_stm32f2xx.c:394:5: error: this 'while' clause does not guard... [-Werror=misleading-indentation]
     while ((RCC->CFGR & (uint32_t)RCC_CFGR_SWS ) != RCC_CFGR_SWS_PLL);
     ^~~~~
MCU/STM32F2xx/SPARK_Firmware_Driver/src/system_stm32f2xx.c:395:5: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'while'
     {
     ^
cc1: all warnings being treated as errors
../build/module.mk:262: recipe for target '../build/target/platform/platform-6-m/./MCU/STM32F2xx/SPARK_Firmware_Driver/src/system_stm32f2xx.o' failed
make[2]: *** [../build/target/platform/platform-6-m/./MCU/STM32F2xx/SPARK_Firmware_Driver/src/system_stm32f2xx.o] Error 1
make[2]: Leaving directory '../Particle_Source/firmware/platform'
../../../build/recurse.mk:11: recipe for target 'platform' failed
make[1]: *** [platform] Error 2
make[1]: Leaving directory '../Particle_Source/firmware/modules/photon/system-part1'
makefile:85: recipe for target '../Particle_Source/firmware/modules/photon/system-part1/makefile' failed
make: *** [../Particle_Source/firmware/modules/photon/system-part1/makefile] Error 2

```
Can you please help me?

cbaser, 2018-06-25:
I need to downgrade my arm-none-eabi-gcc and it solved. Thanks for helping!

closed 2018-06-25 by cbaser