particle-iot/device-os #1556: [Photon/P1] 802.11n only mode not working on Photon/P1

technobly, 2018-07-10:
## Bug Report

802.11n only mode not working on Photon/P1

### Expected Behavior

Photon/P1 devices are built with a Cypress 43362 Wi-Fi radio that is designed to connect to 802.11b/g/n Wi-Fi networks.

### Observed Behavior

Tested on a ASUS RT-N66U and with 2.4GHz *802.11n only mode*, 0.6.0 connects, but 0.7.0-rc.1, 0.7.0 and 0.8.0-rc.8 do not connect.

### Steps to Reproduce

See above.

### Test App

You can use Tinker.

### References

https://community.particle.io/t/p0-wireless-n-only-mode-not-working-on-0-7-0-and-later-firmware/42748

kasperkamperman, 2019-03-29:
Any milestone or plan when this bug will be fixed?

DriftingShadows, 2019-04-01:
any updates on this? still doesn't work on 1.0.1

Bacichetti, 2019-04-17:
Still no update on this issue??
How it's possible after so many months? How do we tell our clients that N mode does not work? It's on the specifications.

kasperkamperman, 2019-04-22:
I'm also surprised that this is not a priority and a solution is included in the release candidate. My fear with the 3rd generation was that the 2nd generation (Photon), would be left alone. Which unfortunately seems the case.

technobly, 2019-04-26:
@kasperkamperman @DriftingShadows @Bacichetti if you have the ability to test #1755 to see if it resolves the issue for y'all, your feedback is appreciated.
```
modules $ make clean all -s PLATFORM=photon COMPILE_LTO=n APP=tinker program-dfu
```

kasperkamperman, 2019-04-27:
Of course I'd like to test. However what are the steps that I have to follow? Do I need to download a certain firmware and then run the make command?

technobly, 2019-04-27:
@kasperkamperman I put more steps on a community post, but I just compiled some binaries for Photon and P1 for you to try. 
 This is compatible with 1.1.0-rc.1 compiled apps, or you can just use the included tinker binary.  You shouldn't need the bootloader, but just in case it's part of the natural build process so I zipped it all up.
https://www.dropbox.com/s/07rusi7uwe262wd/particle_device-os%401.1.0-rc.1-pr-1755.zip?dl=0

closed 2019-04-29 by avtolstoy

Bacichetti, 2019-04-30:
I did test the fix here and it does fix the issue with 802.11n mode.
Thanks!

kasperkamperman, 2019-04-30:
Thanks for the fix!

PR #1755 (2019-04-29): [Photon/P1] Fixes 802.11n mode
### Problem

#1556

### Solution

particle-iot/photon-wiced#32

The PR above resolves and incompatibility in WICED code when handling country codes with the current (newer, with KRACK2 security bugfixes) version of CYW43362 firmware we've introduced in #1420, #1419, #1418.

### Steps to Test

Attempt to connect to 802.11n-only AP. The connection should succeed.

### Example App

N/A

### References

- Closes #1526 
- Closes #1556 
- particle-iot/photon-wiced#32
- [CH17929]

---

### Completeness

- [x] User is totes amazing for contributing!
- [x] Contributor has [... cut]