particle-iot/device-os #1560: am using GNU tools ARM Embedded v7.3.1 with IDE keil uv5.24 in order to compile your code and it compiles just fine but when it comes to the C++ part the linker fails ang give me bad final value

Isawwad, 2018-07-15:
<details>
  <summary><i>submission notes</i></summary>

```
**Important:** Please sanitize/remove any confidential info like usernames, passwords, org names, product names/ids, access tokens, client ids/secrets, or anything else you don't wish to share.

Please choose one of the following templates **Bug Report** or **Feature/Enhancement Request** and delete the other.

You may also delete this submission notes header if you'd like. Thank you for contributing!
```
</details>

## Bug Report

### Expected Behavior

What should it do?

### Observed Behavior

What did it actually do?

### Steps to Reproduce

How did you make it do that, and which system/user firmware versions were tested?

### Test App

```c
void setup() {
  // A minimal test case to reproduce the issue is super helpful!
}

void loop() {

}
```

### References

Links to the Community, Docs, Other Issues, etc..

---

## Feature/Enhancement Request

Please describe your use case in as much detail as possible. Background information and detailing what the feature will enable you to do, is especially helpful.

rickkas7, 2018-07-15:
At this time, the only supported gcc-arm compiler is version 5.3.1 (5-2016-q1-update).

It appears that you are using 7.3.1. It's unlikely that you will be able to get that to work properly.

Isawwad, 2018-07-15:
i tried even to include one cpp file to the c-files  written in it test code when i try to call the constructor of the class i get this messages 
c:/program files (x86)/gnu tools arm embedded/7 2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/thumb/v7-m\libc.a(lib_a-abort.o): In function `abort':
abort.c:(.text.abort+0xa): undefined reference to `_exit'
c:/program files (x86)/gnu tools arm embedded/7 2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/thumb/v7-m\libc.a(lib_a-sbrkr.o): In function `_sbrk_r':
sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk'
c:/program files (x86)/gnu tools arm embedded/7 2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/thumb/v7-m\libc.a(lib_a-signalr.o): In function `_kill_r':
signalr.c:(.text._kill_r+0x10): undefined reference to `_kill'
c:/program files (x86)/gnu tools arm embedded/7 2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/thumb/v7-m\libc.a(lib_a-signalr.o): In function `_getpid_r':
signalr.c:(.text._getpid_r+0x0): undefined reference to `_getpid'
collect2.exe: error: ld returned 1 exit status

ScruffR, 2018-09-23:
This issue should be closed as it's just a duplicate of #1565, but less info.

closed 2020-03-03 by avtolstoy