particle-iot/device-os #1561: Required dependency gcc-arm 5.3.1 not available in homebrew

dokterbob, 2018-07-16:
<details>
  <summary><i>submission notes</i></summary>

```
**Important:** Please sanitize/remove any confidential info like usernames, passwords, org names, product names/ids, access tokens, client ids/secrets, or anything else you don't wish to share.

Please choose one of the following templates **Bug Report** or **Feature/Enhancement Request** and delete the other.

You may also delete this submission notes header if you'd like. Thank you for contributing!
```
</details>

## Bug Report

### Expected Behavior

Instructions in [dependencies](https://github.com/particle-iot/firmware/blob/develop/docs/dependencies.md) should allow all required dependendencies, specifically the required version of gcc-arm to be installed.

### Observed Behavior

The earliest 5.* gcc available is 5.4 - gcc-arm 5.3 is not available.

### Steps to Reproduce

```
$ brew uninstall gcc-arm-none-eabi && brew install gcc-arm-none-eabi-53
Uninstalling /usr/local/Cellar/gcc-arm-none-eabi/20171218... (5,633 files, 466.7MB)
Error: No available formula with the name "gcc-arm-none-eabi-53" 
==> Searching for a previously deleted formula (in the last month)...
Warning: homebrew/core is shallow clone. To get complete history run:
  git -C "$(brew --repo homebrew/core)" fetch --unshallow

Error: No previously deleted formula found.
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
==> Searching taps...
==> Searching taps on GitHub...
Error: No formulae found in taps.
```

### Suggestion

Officially support gcc-arm 5.4 and update documentation.

rickkas7, 2018-07-16:
The gcc-arm local build FAQ includes installation instructions not using Homebrew, which allow 5.3.1 to be installed:

https://docs.particle.io/faq/particle-tools/local-build/

technobly, 2018-07-16:
@dokterbob Thanks for the heads up on this, we really need to maintain our own brew formulas.  In the meantime, I have updated the docs in `develop` to show you how to install gcc 5.3.1 with homebrew: https://github.com/particle-iot/firmware/blob/108538861cd6ddecc12565b0e7662dc4fa3d8c06/docs/dependencies.md

Please let us know if this works for you, so we can close this issue.  Thanks!

technobly, 2018-07-26:
Closing... please re-open if you find the updated Docs not sufficient.

closed 2018-07-26 by technobly