particle-iot/device-os #1565: Building Particle Firmware with Keil uv5.24 IDE.

Isawwad, 2018-08-08:
am using GNU tools ARM Embedded v7.3.1 with IDE keil uv5.24 in order to compile your code and it compiles just fine , using id = electron, when i tried to use threading a linking error just occurred and i dont know why.

linking...
./objects/active_object.o: In function `std::thread::thread<void (&)(ActiveObjectBase*), ActiveObjectBase*>(void (&)(ActiveObjectBase*), ActiveObjectBase*&&)':
c:\program files (x86)\gnu tools arm embedded\7 2018-q2-update\arm-none-eabi\include\c++\7.3.1/thread:126: undefined reference to `std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)())'
./objects/active_object.o: In function `std::thread::_State::_State()':
c:\program files (x86)\gnu tools arm embedded\7 2018-q2-update\arm-none-eabi\include\c++\7.3.1/thread:67: undefined reference to `vtable for std::thread::_State'
./objects/active_object.o: In function `std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (*)(ActiveObjectBase*), ActiveObjectBase*> > >::~_State_impl()':
c:\program files (x86)\gnu tools arm embedded\7 2018-q2-update\arm-none-eabi\include\c++\7.3.1/thread:178: undefined reference to `std::thread::_State::~_State()'
./objects/active_object.o:(.rodata._ZTINSt6thread11_State_implINS_8_InvokerISt5tupleIJPFvP16ActiveObjectBaseES4_EEEEEE[_ZTINSt6thread11_State_implINS_8_InvokerISt5tupleIJPFvP16ActiveObjectBaseES4_EEEEEE]+0x8): undefined reference to `typeinfo for std::thread::_State'
collect2.exe: error: ld returned 1 exit status
".\Objects\Elkrem_Proto_1.elf" - 1 Error(s), 12 Warning(s).
Target not created.
Build Time Elapsed:  00:00:22

PLEASE HELP ME

rickkas7, 2018-08-08:
It's not possible to use the gcc-arm toolchain version 7.3.1 to build Particle firmware at this time.

Currently, the 5.3.1 (5-2016-q1-update) is the supported gcc-arm version.

Prior to Device OS version 0.6.3 you can also use the 4.9-2015-q3-update.

Isawwad, 2018-08-08:
i did as i told and now iave this linking error

linking...
./objects/active_object.o: In function `std::thread::thread<void (&)(ActiveObjectBase*), ActiveObjectBase*>(void (&)(ActiveObjectBase*), ActiveObjectBase*&&)':
c:\program files (x86)\gnu tools arm embedded\5.3 2016q1\arm-none-eabi\include\c++\5.3.1/thread:142: undefined reference to `std::thread::_M_start_thread(std::shared_ptr<std::thread::_Impl_base>)'
collect2.exe: error: ld returned 1 exit status
".\Objects\Elkrem_Proto_1.elf" - 1 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:04

Isawwad, 2018-08-08:
whats the problem

Isawwad, 2018-08-08:
in active_object.cpp file "void ActiveObjectBase::start_thread()" if i comment this function build is done properly why it does a linking error

Isawwad, 2018-08-12:
Update!!
when i remove RTOS files and run super loop in (while (1)) in main the problem disappears and every time i press reset button in keil it returns to main once again so the question is what could be wrong with my RTOS used from your repo

avtolstoy, 2020-03-03:
`std::thread` usage has since been removed.

closed 2020-03-03 by avtolstoy