particle-iot/device-os #1566: Using threads in RTOS

Isawwad, 2018-08-16:
<details>
  <summary><i>submission notes</i></summary>
Hello,
Is there any reason else but the ( memory wise thing )that made toy use threads than tasks in RTOS
```
**Important:** Please sanitize/remove any confidential info like usernames, passwords, org names, product names/ids, access tokens, client ids/secrets, or anything else you don't wish to share.

Please choose one of the following templates **Bug Report** or **Feature/Enhancement Request** and delete the other.

You may also delete this submission notes header if you'd like. Thank you for contributing!
```
</details>

## Bug Report

### Expected Behavior

What should it do?

### Observed Behavior

What did it actually do?

### Steps to Reproduce

How did you make it do that, and which system/user firmware versions were tested?

### Test App

```c
void setup() {
  // A minimal test case to reproduce the issue is super helpful!
}

void loop() {

}
```

### References

Links to the Community, Docs, Other Issues, etc..

---

## Feature/Enhancement Request

Please describe your use case in as much detail as possible. Background information and detailing what the feature will enable you to do, is especially helpful.

beriberikix, 2018-08-16:
Can you expand on your question? Are you asking why we use Threads instead of Tasks?

Isawwad, 2018-08-16:
yes Threads instead of Tasks why? is it the memory optimization or other purpose ?

m-mcgowan, 2018-08-16:
Tasks are what FreeRTOS calls Threads.  They have their own stack, their own execution state and are time-sliced between. They are threads, just with a different name.

rickkas7, 2018-08-16:
This may be helpful if you want to use threads:

https://github.com/rickkas7/particle-threads

m-mcgowan, 2018-08-29:
It looks like the original question has been resolved so I will close this. @Isawwad, should you need further clarification just leave a comment.

closed 2018-08-29 by m-mcgowan