particle-iot/device-os #1569: 0.8.0-rc9 on Core function command string is corrupt

mike4today, 2018-08-21:
<details>
  <summary><i>Firmware 0.8.0-rc9 on Core Particle.function when called the String is corrupt</i></summary>

</details>

## Bug Report
Bug with Particle Core using firmware 0.8.0-rc9
When calling Particle.function the string submitted to the function gets corrupt. 

### Expected Behavior
int x(String command) should be the string submitted.

### Observed Behavior
response for Particle.Publish(command)
{"data":"\u000b\u001ag�Q�[\u001f�x���4�t\u001f����G-�uI��gͶ*\u001d\u000b�~��n�(UFfDFE�\u001d�_�\n~�μV�w�l\u0011�A\u0002��\u0003�f\u0005testFDtest\u000e\u000e\u000e\u000e\u000e\u000e\u000e\u000e\u000e\u000e\u000e\u000e\u000e\u000e","ttl":60,"published_at":"2018-08-21T14:24:46.976Z","coreid":"--removed--","name":"testVal"}

### Steps to Reproduce
Flash 0.8.0-rc9 to Core with example below. From devices call testF with value of "test". testVal will publish what was received and should be "test" as sent, but that is not what is received.
Same test works fine with 0.7.0 on Core. 
Results on 0.7.0 : {"data":"test","ttl":60,"published_at":"2018-08-21T14:27:19.049Z","coreid":"--removed--","name":"testVal"}

### Test App

```c

void setup() {
  Particle.function("testF",      testF);
}

void loop() {

}

int testF(String command){
    Particle.publish("testVal",command,60,PRIVATE);
    if(command=="test"){
        return 1;
    }else{
        return 2;
    }
}

```

rlogiacco, 2019-06-12:
This issue is still affecting versions `1.0.0`, `1.0.1` and `1.1.0` when used in conjuction with an old Core device.

Another valid test to reproduce the issue and excludes any intervention of the `publish` function could be:
```
void setup() {
  Particle.function("testF",      testF);
  Serial.begin(9600);
}

void loop() {

}

int testF(String extra) {
    Serial.println(extra);
}
```

**PLEASE FIX!**

technobly, 2019-06-12:
Please try Device OS 1.2.0-rc.1 or better yet, 1.2.1-rc.2 which also has this fix.
If you don't have enough memory for your application on the Core, you can also try downgrading to 0.7.0, 0.6.3, etc.. for Core.

closed 2019-06-12 by technobly