particle-iot/device-os #1572: Error Building Firmware v0.8.0-rc.10

kcox920, 2018-09-14:
<details>
  <summary><i>Error Building Firmware v0.8.0-rc.10</i></summary>

Checked out the v0.8.0-rc.10 tree and attempted to build. However, the build failed with the following error:
“
Building cpp file: src/spark_protocol_functions.cpp
Invoking: ARM GCC CPP Compiler
mkdir -p …/build/target/communication/platform-10-m-prod-10/src/
arm-none-eabi-gcc -I/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1/include -DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_THREADING=1 -DPLATFORM_ID=10 -DPLATFORM_NAME=electron -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD00A -DUSBD_PID_CDC=0xC00A -DSYSTEM_VERSION_STRING=0.8.0-rc.10 -DRELEASE_BUILD -Werror -I. -MD -MP -MF /home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/makefile.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb -DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_THREADING=1 -DPLATFORM_ID=10 -DPLATFORM_NAME=electron -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD00A -DUSBD_PID_CDC=0xC00A -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb -DPRODUCT_ID=10 -DPRODUCT_FIRMWARE_VERSION=65535 -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -DMBEDTLS_CONFIG_FILE=”<mbedtls_config.h>" -fno-builtin -DSYSTEM_VERSION_STRING=0.8.0-rc.10 -DRELEASE_BUILD -Werror -I…/…/…/modules/electron/system-part2/inc -I…/…/…/modules/shared/stm32f2xx/inc -I…/…/…/communication/src -I…/…/…/services/inc -I…/…/…/services/nanopb -I…/…/…/platform/shared/inc -I…/…/…/platform/MCU/STM32F2xx/STM32_USB_Host_Driver/inc -I…/…/…/platform/MCU/STM32F2xx/STM32_StdPeriph_Driver/inc -I…/…/…/platform/MCU/STM32F2xx/STM32_USB_OTG_Driver/inc -I…/…/…/platform/MCU/STM32F2xx/STM32_USB_Device_Driver/inc -I…/…/…/platform/MCU/STM32F2xx/SPARK_Firmware_Driver/inc -I…/…/…/platform/MCU/shared/STM32/inc -I…/…/…/platform/MCU/STM32F2xx/CMSIS/Include -I…/…/…/platform/MCU/STM32F2xx/CMSIS/Device/ST/Include -I…/…/…/crypto/mbedtls/include -I…/…/…/crypto/inc -I…/…/…/dynalib/inc -I…/…/…/rt-dynalib/inc -I…/…/…/modules/shared/stm32f2xx/inc -I. -MD -MP -MF communication.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -fno-builtin-malloc -fno-builtin-free -fno-builtin-realloc -DUSER_FIRMWARE_IMAGE_SIZE=0x20000 -DUSER_FIRMWARE_IMAGE_LOCATION=0x8080000 -DMODULAR_FIRMWARE=1 -DMODULE_VERSION=309 -DMODULE_FUNCTION=4 -DMODULE_INDEX=1 -DMODULE_DEPENDENCY=4,3,309 -DMODULE_DEPENDENCY2=0,0,0 -D_WINSOCK_H -D_GNU_SOURCE -DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_THREADING=1 -DPLATFORM_ID=10 -DPLATFORM_NAME=electron -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD00A -DUSBD_PID_CDC=0xC00A -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb -DPRODUCT_ID=10 -DPRODUCT_FIRMWARE_VERSION=65535 -DMBEDTLS_CONFIG_FILE="<mbedtls_config.h>" -DSYSTEM_VERSION_STRING=0.8.0-rc.10 -DRELEASE_BUILD -Werror -I./src -I…/hal/inc -I…/hal/shared -I/rtos/FreeRTOSv8.2.2/FreeRTOS/Source/include -I/rtos/FreeRTOSv8.2.2/FreeRTOS/Source/portable/GCC/ARM_CM3 -I…/hal/src/electron -I…/hal/src/stm32f2xx -I…/hal/src/stm32 -I…/dynalib/inc -I…/services/inc -I…/services/nanopb -I…/wiring/inc -I…/crypto/mbedtls/include -I…/crypto/inc -I. -MD -MP -MF …/build/target/communication/platform-10-m-prod-10/src/spark_protocol_functions.o.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -DUSER_FIRMWARE_IMAGE_SIZE=0x20000 -DUSER_FIRMWARE_IMAGE_LOCATION=0x8080000 -DMODULAR_FIRMWARE=1 -DMODULE_VERSION=309 -DMODULE_FUNCTION=4 -DMODULE_INDEX=1 -DMODULE_DEPENDENCY=4,3,309 -DMODULE_DEPENDENCY2=0,0,0 -D_WINSOCK_H -D_GNU_SOURCE -DLOG_MODULE_CATEGORY="“comm”" -fno-exceptions -fno-rtti -fcheck-new -std=gnu++11 -c -o …/build/target/communication/platform-10-m-prod-10/src/spark_protocol_functions.o src/spark_protocol_functions.cpp
In file included from …/hal/inc/interrupts_hal.h:33:0,
from …/wiring/inc/spark_wiring_interrupts.h:31,
from …/wiring/inc/spark_wiring_diagnostics.h:21,
from src/communication_diagnostic.h:1,
from src/publisher.h:28,
from src/protocol.h:12,
from src/lightssl_protocol.h:31,
from src/spark_protocol_functions.cpp:77:
…/hal/src/stm32/interrupts_irq.h:22:23: fatal error: stm32f2xx.h: No such file or directory
"
</details>

## Bug Report
Build failure
### Expected Behavior
Build.
### Observed Behavior
Build failed. 

### Steps to Reproduce
do a fresh checkout of branch "release/v0.8.0-rc.10"
cd modules
make all PLATFORM=electron

(note: observed same issue with branch "release/v0.7.0")

### References
also put this in the community forum:
https://community.particle.io/t/error-building-firmware-v0-8-0-rc-10/43998

attached full build output
[20180914_build_output.txt](https://github.com/particle-iot/firmware/files/2384900/20180914_build_output.txt)

---

kcox920, 2018-09-18:
The following fixes the issue with the build system:
diff --git a/modules/electron/system-part1/makefile b/modules/electron/system-part1/makefile
index 96800fe..4facfee 100644
--- a/modules/electron/system-part1/makefile
+++ b/modules/electron/system-part1/makefile
@@ -1,6 +1,6 @@
 MODULE=system-part1
 WIFI_SYSTEM_MODULE_PATH=.
-PROJECT_ROOT=../../..
+PROJECT_ROOT=$(CURDIR)/../../..
 BUILD_PATH_EXT = $(BUILD_TARGET_PLATFORM)
 HAL_LINK :=
 PLATFORM_DFU = 0x8060000
diff --git a/modules/electron/system-part2/makefile b/modules/electron/system-part2/makefile
index 6031830..3ddc5ae 100644
--- a/modules/electron/system-part2/makefile
+++ b/modules/electron/system-part2/makefile
@@ -1,6 +1,6 @@
 MODULE=system-part2
 WIFI_SYSTEM_MODULE_PATH=.
-PROJECT_ROOT=../../..
+PROJECT_ROOT=$(CURDIR)/../../..
 BUILD_PATH_EXT = $(BUILD_TARGET_PLATFORM)
 HAL_LINK :=
 PLATFORM_DFU = 0x8020000
diff --git a/modules/electron/system-part3/makefile b/modules/electron/system-part3/makefile
index 6610c95..6560a83 100644
--- a/modules/electron/system-part3/makefile
+++ b/modules/electron/system-part3/makefile
@@ -1,6 +1,6 @@
 MODULE=system-part3
 SYSTEM_PART2_MODULE_PATH=.
-PROJECT_ROOT = ../../..
+PROJECT_ROOT = $(CURDIR)/../../..
 BUILD_PATH_EXT = $(BUILD_TARGET_PLATFORM)
 HAL_LINK :=
 PLATFORM_DFU = 0x8040000
diff --git a/modules/electron/user-part/makefile b/modules/electron/user-part/makefile
index 108b23b..b36ed4e 100644
--- a/modules/electron/user-part/makefile
+++ b/modules/electron/user-part/makefile
@@ -1,6 +1,6 @@
 MODULE=user-part
 USER_PART_MODULE_PATH=.
-PROJECT_ROOT=../../..
+PROJECT_ROOT=$(CURDIR)/../../..
 BUILD_PATH_EXT = $(BUILD_TARGET_PLATFORM)
 HAL_LINK :=
 PLATFORM_DFU = $(USER_FIRMWARE_IMAGE_LOCATION)
--- a/modules/photon/system-part1/makefile
+++ b/modules/photon/system-part1/makefile
@@ -1,6 +1,6 @@
 MODULE=system-part1
 WIFI_SYSTEM_MODULE_PATH=.
-PROJECT_ROOT=../../..
+PROJECT_ROOT=$(CURDIR)/../../..
 BUILD_PATH_EXT = $(BUILD_TARGET_PLATFORM)
 HAL_LINK :=
 PLATFORM_DFU = 0x8020000
diff --git a/modules/photon/system-part2/makefile b/modules/photon/system-part2/makefile
index 56340c5..b856177 100644
--- a/modules/photon/system-part2/makefile
+++ b/modules/photon/system-part2/makefile
@@ -1,6 +1,6 @@
 MODULE=system-part2
 SYSTEM_PART2_MODULE_PATH=.
-PROJECT_ROOT = ../../..
+PROJECT_ROOT = $(CURDIR)/../../..
 BUILD_PATH_EXT = $(BUILD_TARGET_PLATFORM)
 HAL_LINK :=
 PLATFORM_DFU = 0x8060000
diff --git a/modules/photon/user-part/makefile b/modules/photon/user-part/makefile
index e9a85ba..2a69d96 100644
--- a/modules/photon/user-part/makefile
+++ b/modules/photon/user-part/makefile
@@ -1,6 +1,6 @@
 MODULE=user-part
 USER_PART_MODULE_PATH=.
-PROJECT_ROOT=../../..
+PROJECT_ROOT=$(CURDIR)/../../..
 BUILD_PATH_EXT = $(BUILD_TARGET_PLATFORM)
 HAL_LINK :=
 PLATFORM_DFU = $(USER_FIRMWARE_IMAGE_LOCATION)

kcox920, 2018-09-18:
However, in fixing the build system this has exposed a typedef conflict. The last error is below, but I've also attached the full build log as a file:
Building cpp file: src/communication_dynalib.cpp
Invoking: ARM GCC CPP Compiler
mkdir -p ../build/target/communication/platform-10-m-prod-10/src/
arm-none-eabi-gcc -I/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1/include -DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_THREADING=1 -DPLATFORM_ID=10 -DPLATFORM_NAME=electron -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD00A -DUSBD_PID_CDC=0xC00A -DSYSTEM_VERSION_STRING=0.8.0-rc.10 -DRELEASE_BUILD -Werror  -I. -MD -MP -MF /home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/makefile.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb -DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_THREADING=1 -DPLATFORM_ID=10 -DPLATFORM_NAME=electron -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD00A -DUSBD_PID_CDC=0xC00A -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb -DPRODUCT_ID=10 -DPRODUCT_FIRMWARE_VERSION=65535 -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -DMBEDTLS_CONFIG_FILE="<mbedtls_config.h>" -fno-builtin -DSYSTEM_VERSION_STRING=0.8.0-rc.10 -DRELEASE_BUILD -Werror -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../modules/electron/system-part2/inc -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../modules/shared/stm32f2xx/inc -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../communication/src -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../services/inc -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../services/nanopb -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../platform/shared/inc -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../platform/MCU/STM32F2xx/STM32_USB_Host_Driver/inc -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../platform/MCU/STM32F2xx/STM32_StdPeriph_Driver/inc -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../platform/MCU/STM32F2xx/STM32_USB_OTG_Driver/inc -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../platform/MCU/STM32F2xx/STM32_USB_Device_Driver/inc -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../platform/MCU/STM32F2xx/SPARK_Firmware_Driver/inc -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../platform/MCU/shared/STM32/inc -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../platform/MCU/STM32F2xx/CMSIS/Include -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../platform/MCU/STM32F2xx/CMSIS/Device/ST/Include -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../crypto/mbedtls/include -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../crypto/inc -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../dynalib/inc -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../rt-dynalib/inc -I/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../modules/shared/stm32f2xx/inc -I. -MD -MP -MF communication.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -fno-builtin-malloc -fno-builtin-free -fno-builtin-realloc -DUSER_FIRMWARE_IMAGE_SIZE=0x20000 -DUSER_FIRMWARE_IMAGE_LOCATION=0x8080000 -DMODULAR_FIRMWARE=1 -DMODULE_VERSION=309 -DMODULE_FUNCTION=4 -DMODULE_INDEX=1 -DMODULE_DEPENDENCY=4,3,309 -DMODULE_DEPENDENCY2=0,0,0 -D_WINSOCK_H -D_GNU_SOURCE -DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_THREADING=1 -DPLATFORM_ID=10 -DPLATFORM_NAME=electron -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD00A -DUSBD_PID_CDC=0xC00A -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb -DPRODUCT_ID=10 -DPRODUCT_FIRMWARE_VERSION=65535 -DMBEDTLS_CONFIG_FILE="<mbedtls_config.h>" -DSYSTEM_VERSION_STRING=0.8.0-rc.10 -DRELEASE_BUILD -Werror -I./src -I../hal/inc -I../hal/shared -I/rtos/FreeRTOSv8.2.2/FreeRTOS/Source/include -I/rtos/FreeRTOSv8.2.2/FreeRTOS/Source/portable/GCC/ARM_CM3 -I../hal/src/electron -I../hal/src/stm32f2xx -I../hal/src/stm32 -I../dynalib/inc -I../services/inc -I../services/nanopb -I../wiring/inc -I../crypto/mbedtls/include -I../crypto/inc -I. -MD -MP -MF ../build/target/communication/platform-10-m-prod-10/src/communication_dynalib.o.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -DUSER_FIRMWARE_IMAGE_SIZE=0x20000 -DUSER_FIRMWARE_IMAGE_LOCATION=0x8080000 -DMODULAR_FIRMWARE=1 -DMODULE_VERSION=309 -DMODULE_FUNCTION=4 -DMODUL [... cut ...]
/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/../../../build/recurse.mk:11: recipe for target 'communication' failed
make[1]: *** [communication] Error 2
make[1]: Leaving directory '/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2'
makefile:85: recipe for target '/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/makefile' failed
make: *** [/home/kcox/dev/fstudio/projects/os_particle/firmware/modules/electron/system-part2/makefile] Error 2

rickkas7, 2018-09-19:
I wasn't able to reproduce this error. I tested on Ubuntu Linux 16.04.

```
$ uname -a
Linux ubuntu 4.4.0-31-generic #50-Ubuntu SMP Wed Jul 13 00:07:12 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

$ arm-none-eabi-gcc --versionarm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.3.1 20160307 (release) [ARM/embedded-5-branch revision 234589]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ make --version
GNU Make 4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

```
rickk@ubuntu:~/Documents$ git clone https://github.com/particle-iot/firmware.git
rickk@ubuntu:~/Documents$ cd firmware
rickk@ubuntu:~/Documents/firmware$ git checkout release/v0.8.0-rc.10
rickk@ubuntu:~/Documents/firmware$ cd modules
rickk@ubuntu:~/Documents/firmware/modules$ make all PLATFORM=electron
```

The build completed successfully, all three system parts and the user part (Tinker).

kcox920, 2018-09-19:
Ok, finally figured it out.  Normally for cross compilation projects I use an environment setup script like below:
***********************************************
kcox:modules$ cat ../../../../setupCrossEnv.sh 
#!/bin/sh

DIR=/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1
ARM=arm-none-eabi
export AR=$DIR/bin/$ARM-ar
export CC=$DIR/bin/$ARM-gcc
export LD=$DIR/bin/$ARM-ld
export NM=$DIR/bin/$ARM-nm
export RANLIB=$DIR/bin/$ARM-ranlib
export STRIP=$DIR/bin/$ARM-strip
export CFLAGS="-I$DIR/include"
export CROSS_COMPILE=$DIR/bin/$ARM-
export ARCH=arm

#export LDFLAGS="--accept-unknown-input-arch --verbose"

export PATH=$PATH:$DIR/bin
***********************************************

This led to the environment being set to the following (with regard to "arm")
************************************
kcox:firmware$ env | grep arm
AR=/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1/bin/arm-none-eabi-ar
NM=/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1/bin/arm-none-eabi-nm
PATH=/usr/local/cuda-7.0/bin:/home/kcox/bin:/usr/local/bin:/usr/local/sbin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/kcox/bin:/opt:/home/kcox/dev/android-sdk-linux/tools:/home/kcox/dev/android-sdk-linux/platform-tools:/home/kcox/dev/android-ndk-r16b:/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1/bin
LD=/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1/bin/arm-none-eabi-ld
STRIP=/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1/bin/arm-none-eabi-strip
CFLAGS=-I/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1/include
ARCH=arm
RANLIB=/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1/bin/arm-none-eabi-ranlib
CROSS_COMPILE=/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1/bin/arm-none-eabi-
CC=/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1/bin/arm-none-eabi-gcc
kcox:firmware$ env | grep LDFLAGS
kcox:firmware$
*****************************************

That environment setup causes the build to break.  Eliminating all of the exports from the script except the updated path such as
****************************************
kcox:modules$ cat ../../../../setupCrossEnv.sh 
#!/bin/sh

DIR=/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1
#ARM=arm-none-eabi
#export AR=$DIR/bin/$ARM-ar
#export CC=$DIR/bin/$ARM-gcc
#export LD=$DIR/bin/$ARM-ld
#export NM=$DIR/bin/$ARM-nm
#export RANLIB=$DIR/bin/$ARM-ranlib
#export STRIP=$DIR/bin/$ARM-strip
#export CFLAGS="-I$DIR/include"
#export CROSS_COMPILE=$DIR/bin/$ARM-
#export ARCH=arm

#export LDFLAGS="--accept-unknown-input-arch --verbose"

export PATH=$PATH:$DIR/bin
*******************************************

The environment with regard to arm is now:
**************************************************
kcox:modules$ env | grep arm
PATH=/usr/local/cuda-7.0/bin:/home/kcox/bin:/usr/local/bin:/usr/local/sbin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/kcox/bin:/opt:/home/kcox/dev/android-sdk-linux/tools:/home/kcox/dev/android-sdk-linux/platform-tools:/home/kcox/dev/android-ndk-r16b:/home/kcox/dev/compilers/gcc-arm-none-eabi-5_3-2016q1/bin
**************************************************

And now the build completes without error. Very weird, I've never seen a cross compilation failure like this before (and I've been doing this for years...). There is something peculiar about how this build system was put together for sure. Nonetheless I'm going to skip figuring out why, as things build fine now and it's off to other tasks.

If you try running my environment script (corrected of course for where your compiler is) you should get the same error.

closed 2018-09-19 by kcox920