particle-iot/device-os #1574: SYSTEM_MODE(MANUAL) does not perform as described in manual in v0.8.x

rtpavlovsk21, 2018-09-18:
<details>
  <summary><i>Code using SYSTEM_MODE(MANUAL) working in 0.6.x but not in 0.8.x</i></summary>

</details>

## Bug Report

### Expected Behavior
Expect SYSTEM_MODE(MANUAL) to not attempt to automatically connect to the cloud (application needs to run before connected).
### Observed Behavior
Code that ran in 0.6.x ran without connecting to cloud (white breathing led). While same code in 0.8.x flashes green led. 
### Steps to Reproduce
0.6.x works
0.8.x does not (specifically 0.8.0-rc.3 0.8.0-rc.10)
arm-none-eabi-gcc 5.3.1
### Test App

```c
#include "application.h"
#include "math.h"
                                                                
SYSTEM_MODE(MANUAL);

void setup()
{
}

void loop(){
}
```

rickkas7, 2018-09-19:
I was unable to reproduce this. Could you attach the output from:

```
particle serial inspect
```

with the device connected by USB and in listening mode (blinking dark blue).

It's possible that it's going into blinking green because there's an unmet dependency, most likely the bootloader if you manually flashed system-part1-0.8.0-rc.10-photon.bin and system-part2-0.8.0-rc.10-photon.bin but did not flash the bootloader. It's going into blinking green because it needs to get into safe mode to download the 0.8.0 bootloader.

rtpavlovsk21, 2018-09-19:
Hey Rikkas,
This seems to be the issue, I started installing the Particle CLI but also connected the photon to the internet while I was doing that... and it appears that the function is now behaving as I would expect in the v0.8.x. firmware. Thank you for your help,

closed 2018-09-19 by rtpavlovsk21