particle-iot/device-os #1575: can a device publish at the rate of more than 4 event/sec?

jinyangqiao, 2018-09-21:
## Feature/Enhancement Request

Hi,
Now I have many smart devices(such as light) that belongs to a P1，
because there are more smart devices and I need to get information about these devices frequently，
then send it to the cloud via particle.publish. So 4 event/sec can't meet my needs at all.
What I want to know is, is there any way to make device can publish at the rate of more than 4 event/sec?

My message is like this:
my smart devices(by RF2.4) --> P1(by publish) --> my cloud

Thanks very much!

rickkas7, 2018-09-21:
No, the limit cannot be changed. And it's actually 4 in a second, followed by 4 seconds of not sending publishes, so the limit is really 1 per second continuously.

The alternatives are:

- Combine multiple things into a single publish. This is even easier with 0.8.0 as the limit is 622 bytes per publish vs. 255.
- Use technique other than publish, such as a direct TCP connection to your server.

jinyangqiao, 2018-09-25:
Ok, I see and I will try like this.
Thanks very much!

closed 2018-09-25 by jinyangqiao