particle-iot/device-os #1592: E402 module wireless connection resets after calling client.stop

wesner0019, 2018-11-07:
## Bug Report
After calling client.stop with an E402 LTE m1 module in semi-automatic mode with OS 0.8.0-rc.11 the module blocks user code as it appears to reset the network.  Please see the below post for more details. 

### Expected Behavior
It should not reset the Cloud/cellular connection

### References
https://community.particle.io/t/lte-gets-blocked-after-client-stop/45011

wesner0019, 2018-11-07:
This may be related to this bug too:
https://github.com/particle-iot/firmware/issues/1512

hine0088, 2019-02-01:
Below is the section where it is blocking and restarting. 
```
|2019-02-01 10:54:27|   369.407 AT read OK    6 "\r\nOK\r\n"|
|2019-02-01 10:54:27|socketClose(1)|
|2019-02-01 10:54:27|   369.418 AT send      12 "AT+USOCL=1\r\n"|
|2019-02-01 10:54:37|socketFree(1)|
|2019-02-01 10:54:38|0000379930 [comm.coap] TRACE: sending message id=105|
|2019-02-01 10:54:38|0000379930 [system] TRACE: send 39|
|2019-02-01 10:54:38|socketSendTo(0,18.206.150.141,5684,,39)|
|2019-02-01 10:54:38|   379.927 AT send      37 "AT+USOST=0,\"18.206.150.141\",5684,39\r\n"|
|2019-02-01 10:54:48|   389.933 AT send      37 "AT+USOST=0,\"18.206.150.141\",5684,39\r\n"|
|2019-02-01 10:54:58|mbedtls/library/ssl_tls.c:2971: mbedtls_ssl_flush_output() returned -1 (-0x0001)|
|2019-02-01 10:54:58|mbedtls/library/ssl_tls.c:7165: mbedtls_ssl_write_record() returned -1 (-0x0001)|
|2019-02-01 10:54:58|   399.937 AT send      10 "AT+CCLK?\r\n"|
```

```
|2019-02-01 10:54:26|Socket 1: handle 1 was created|
|---|---|
|2019-02-01 10:54:26|0000368424 [wiring] TRACE: TCPClient socket=1|
|2019-02-01 10:54:26|socketConnect(1,port:8000)|
|2019-02-01 10:54:26|   368.422 AT send      33 "AT+USOCO=1,\"1""\",8000\r\n"|
|2019-02-01 10:54:27|   368.573 AT read OK    6 "\r\nOK\r\n"|
|2019-02-01 10:54:27|   369.025 AT send      15 "AT+USOWR=1,54\r\n"|
|2019-02-01 10:54:27|   369.035 AT read  >    3 "\r\n@"|
|2019-02-01 10:54:27|   369.085 AT send      54 "\x85M\xbd@\x00\x008\xbc\xcb\xe7\x00\x00\b+\t\x00\x00\x00\xff\xe0\xf5\x05\x00\x00/\xe6\x02\x00\x00\x00L\x04\x00\x00\x00\x00\xc9\xb1\b\x19\x00\x00\xa9>\x1e\x00\x00\x00\xfc] \x00\x00\x00"|
|2019-02-01 10:54:27|   369.097 AT read  +   16 "\r\n+USOWR: 1,54\r\n"|
|2019-02-01 10:54:27|   369.107 AT read OK    6 "\r\nOK\r\n"|
|2019-02-01 10:54:27|   369.107 AT send      14 "AT+USORD=1,0\r\n"|
|2019-02-01 10:54:27|   369.118 AT read UNK   1 "\r"|
|2019-02-01 10:54:27|   369.128 AT read UNK   5 "\r\n\r\r\n"|
|2019-02-01 10:54:27|   369.138 AT read UNK  11 "+USORD: 1,0"|
|2019-02-01 10:54:27|   369.148 AT read OK    6 "\r\nOK\r\n"|
|2019-02-01 10:54:27|   369.210 AT send      16 "AT+USOWR=1,618\r\n"|
|2019-02-01 10:54:27|   369.221 AT read  >    3 "\r\n@"|
|2019-02-01 10:54:27|   369.271 AT send     618 "\x85M\xbd@\x00\x00\x88(vX\a\x00\x88\f\x82\xac\t\x00\x88(vX\a\x00n\x0e\x82\xac\t\x00q*vX\a\x00r\x0e\x82\xac\t\x00s*vX\a\x00v\x0e\x82\xac\t\x00w*vX\a\x00\x7f\x0e\x82\xac\t\x00\x7f*vX\a\x00\x81\x0e\x82\xac\t\x00\x82*vX\a\x00\x83\x0e\x82\xac\t\x00\x83*vX\a\x00\xa5\x8b\x8c<\xd4\x00\xa5\xd7\x99T\x0e\x00\xa5\xa7\x87\xe5<\x00\xa5[z\xcd\x02\x01\xa5K\xd5\xf8\x19\x00\xcc\xaf\xcf\xddJ\x00\xf2\xaf\xcf\xddJ\x007\xb0\xcf\xddJ\x00\x98\xb0\xcf\xddJ\x00s\xb1\xcf\xddJ\x00D\xb2\xcf\xddJ\x00\x16\xb8\xcf\xddJ\x00\xa2\xb8\xcf\xddJ\x00\x06\xb9\xcf\xddJ\x00\xf9\xbc\xcf\xddJ\x00j\xbf\xcf\xddJ\x00\xa8\xd1\xcf\xddJ\x00\xe9\xe4\x82\xac\t\x00\xf3\xe4\x82\xac\t\x00\xf3\xd4\xcf\xddJ\x00\xfe\x00wX\a\x00e\xe8\x82\xac\t\x00f\x04wX\a\x00T\xe9\x82\xac\t\x00s\x05wX\a\x00\xde\xe9\x82\xac\t\x00\xee\xa1\xe7\x85O\x00\xeeQoy\x00\x01\xee\x99\x11\x9d%\x00\xee\xc18,\x8c\x01\xee\xb9bCb\x01\xee\x9dn\x97d\x01\xeem\\(\x93\x01\x00~\x1d\xf1'\x00\x00\x06wX\a\x00=\xea\x82\xac\t\x00N\xa2\xe7\x85O\x00N\x1a\x87!\x05\x01NJ\x92\x93k\x01O\x9a\x11\x9d%\x00O\xc28,\x8c\x01O\xbabCb\x01O\x8aP\xd4\x90\x01Sj\xff-T\x00S\x86\xf3\xd9Q\x00Y~\x1d\xf1'\x00Y\x06wX\a\x00\x9d\xf8\xc3\x89H\x00\x15\xd6\x9f\xab\xa5\x000\xf2\xa6\xa8\x10\x000\xaa\x04\x85\xeb\x000\x92{\xcd\x02\x010\xfan\x97d\x01\xe5\xa8\x94W\xa3\x00\xb4\xa6Szd\x00\xb8j\x00.T\x00\xb8B\xcb\xa4\x17\x00\xb8:\xe7\xc1\x17\x01\xb8\x82{\xebf\x01\xd0\x86\xf4\xd9Q\x00\xd1\xfa/~]\x00\x01\x95\x82\xe8\xd1\x00\x01\xc5\x9bT\x0e\x00\x01epy\x00\x01\x019\xd7\xf8\x19\x00\x8d\xf4\xd0\xddJ\x00\xb1\xf4\xd0\xddJ\x00<\x06\x84\xac\t\x00<\"xX\a\x00C\x06\x84\xac\t\x00M\x06\x84\xac\t\x00M\xf6\xd0\xddJ\x00W\"xX\a\x00k\x06\x84\xac\t\x00l\"xX\a\x00l\x06\x84\xac\t\x00m\"xX\a\x00n\x06\x84\xac\t\x00o\"xX\a\x00o\x06\x84\xac\t\x00p\"xX\a\x00\xc8\x13\xc5\x89H\x007\b\x84\xac\t\x00O\b\x84\xac\t\x00Z\xf8\xd0\xddJ\x00g$xX\a\x00\xa8\xe9\x0f\x15\xb6\x01"|
|2019-02-01 10:54:27|   369.356 AT read  +   17 "\r\n+USOWR: 1,618\r\n"|
|2019-02-01 10:54:27|   369.366 AT read OK    6 "\r\nOK\r\n"|
|2019-02-01 10:54:27|   369.366 AT send      14 "AT+USORD=1,0\r\n"|
|2019-02-01 10:54:27|   369.377 AT read UNK   1 "\r"|
|2019-02-01 10:54:27|   369.387 AT read UNK   5 "\r\n\r\r\n"|
|2019-02-01 10:54:27|   369.397 AT read UNK  11 "+USORD: 1,0"|
|2019-02-01 10:54:27|   369.407 AT read OK    6 "\r\nOK\r\n"|
|2019-02-01 10:54:27|socketClose(1)|
|2019-02-01 10:54:27|   369.418 AT send      12 "AT+USOCL=1\r\n"|
|2019-02-01 10:54:37|socketFree(1)|
|2019-02-01 10:54:38|0000379930 [comm.coap] TRACE: sending message id=105|
|2019-02-01 10:54:38|0000379930 [system] TRACE: send 39|
|2019-02-01 10:54:38|socketSendTo(0,18.206.150.141,5684,,39)|
|2019-02-01 10:54:38|   379.927 AT send      37 "AT+USOST=0,\"18.206.150.141\",5684,39\r\n"|
|2019-02-01 10:54:48|   389.933 AT send      37 "AT+USOST=0,\"18.206.150.141\",5684,39\r\n"|
|2019-02-01 10:54:58|mbedtls/library/ssl_tls.c:2971: mbedtls_ssl_flush_output() returned -1 (-0x0001)|
|2019-02-01 10:54:58|mbedtls/library/ssl_tls.c:7165: mbedtls_ssl_write_record() returned -1 (-0x0001)|
|2019-02-01 10:54:58|   399.937 AT send      10 "AT+CCLK?\r\n"|
|2019-02-01 10:54:58|0000399940 [comm.protocol] ERROR: Event loop error 3|
|2019-02-01 10:54:58|0000399941 [system] WARN: Communication loop error, closing cloud socket|
[... cut ...]
|2019-02-01 10:54:58|   400.069 AT send      15 "AT+USOCTL=0,1\r\n"|
|2019-02-01 10:54:58|   400.079 AT read  +   19 "\r\n+USOCTL: 0,1,11\r\n"|
|2019-02-01 10:54:58|   400.089 AT read OK    6 "\r\nOK\r\n"|
|2019-02-01 10:54:58|Socket handle 0 was open, now closing...|
|2019-02-01 10:54:58|   400.089 AT send      12 "AT+USOCL=0\r\n"|
|2019-02-01 10:54:58|   400.101 AT read ERR  37 "\r\n+CME ERROR: Operation not allowed\r\n"|
|2019-02-01 10:54:58|   400.103 AT send      15 "AT+USOCTL=1,1\r\n"|
```

technobly, 2019-03-27:
This looks like the issue that was resolved in v1.0.1.  When cellular signal is lost, the USOST and USOWR commands can cause the AT interface to become unresponsive.  This can still occur in v1.0.1, but the issue is better mitigated and recovery is much faster.  See https://github.com/particle-iot/device-os/pull/1689

closed 2019-03-27 by technobly