particle-iot/device-os #1605: User thread and os timer freeze,Photon System Firmware 0.5.4

aeris-ming, 2018-11-23:
issue description:
1.User thread/OS timer freeze random,RGB led stop blink too
2.When device freeze,external I/O interruption can work

st-link debug:
1.connect st-link and wait for freeze
2.after device freeze,unlucky back trace can not work
CPU states:
```
> poll
background polling: on
TAP: stm32f2x.cpu (enabled)
target state: running

```
Register value:
```
> reg pc
pc (/32): 0x08075BBE
> reg psp
psp (/32): 0x20005014
> reg lr
lr (/32): 0x08076823
> reg r0
r0 (/32): 0x20004AF4
> reg r1
r1 (/32): 0x20004D30
> reg r2
r2 (/32): 0x00000008
> reg r3
r3 (/32): 0x20004D30
> reg r4
r4 (/32): 0x20004D30
> reg r5
r5 (/32): 0x00000008
> reg r6
r6 (/32): 0x00000000
> reg r7
r7 (/32): 0x00000001
> reg r8
r8 (/32): 0x00000000
> reg r9
r9 (/32): 0x00000000
> reg r10
r10 (/32): 0x20017A18
> reg r11
r11 (/32): 0x01C3B2F3
> reg r12
r12 (/32): 0x00000007
```

Locate PC:
```
08075bae <vListInsert>:
 8075bae:       680a            ldr     r2, [r1, #0]
 8075bb0:       b530            push    {r4, r5, lr}
 8075bb2:       1c53            adds    r3, r2, #1
 8075bb4:       d101            bne.n   8075bba <vListInsert+0xc>
 8075bb6:       6903            ldr     r3, [r0, #16]
 8075bb8:       e007            b.n     8075bca <vListInsert+0x1c>
 8075bba:       f100 0308       add.w   r3, r0, #8
 8075bbe:       685c            ldr     r4, [r3, #4]
 8075bc0:       6825            ldr     r5, [r4, #0]
 8075bc2:       4295            cmp     r5, r2
 8075bc4:       d801            bhi.n   8075bca <vListInsert+0x1c>
 8075bc6:       4623            mov     r3, r4
 8075bc8:       e7f9            b.n     8075bbe <vListInsert+0x10>
 8075bca:       685a            ldr     r2, [r3, #4]
 8075bcc:       604a            str     r2, [r1, #4]
 8075bce:       6091            str     r1, [r2, #8]
[... 187 lines cut ...]
 8075bb6:       6903            ldr     r3, [r0, #16]
 8075bb8:       e007            b.n     8075bca <vListInsert+0x1c>
 8075bba:       f100 0308       add.w   r3, r0, #8
 8075bbe:       685c            ldr     r4, [r3, #4]
 8075bc0:       6825            ldr     r5, [r4, #0]
 8075bc2:       4295            cmp     r5, r2
 8075bc4:       d801            bhi.n   8075bca <vListInsert+0x1c>
 8075bc6:       4623            mov     r3, r4
 8075bc8:       e7f9            b.n     8075bbe <vListInsert+0x10>
 8075bca:       685a            ldr     r2, [r3, #4]
 8075bcc:       604a            str     r2, [r1, #4]
 8075bce:       6091            str     r1, [r2, #8]
 8075bd0:       608b            str     r3, [r1, #8]
 8075bd2:       6059            str     r1, [r3, #4]
 8075bd4:       6803            ldr     r3, [r0, #0]
 8075bd6:       6108            str     r0, [r1, #16]
 8075bd8:       3301            adds    r3, #1
 8075bda:       6003            str     r3, [r0, #0]
 8075bdc:       bd30            pop     {r4, r5, pc}
```
What can be the reason?
https://www.freertos.org/FreeRTOS_Support_Forum_Archive/November_2014/freertos_Crash_in_vListInsert_due_to_xValueOfInsertion_is_equal_to_pxNext-_xItemValue_3d5acf6cj.html

How to reproduction this issue?
What I can do now is wait hours maybe days.I will try my best to find out.

aeris-ming, 2018-11-23:
I think it is because I call some API in the wrong way.But read the source code of system part is too hard for me.which API may cause this issue?

aeris-ming, 2018-11-23:
just find delay() is not delay.Are you guys sure  Spark_Idle_Events(true); is multi thread safe?

rickkas7, 2018-11-23:
Are you using a software timer (Timer class) or a hardware timer (like SparkIntervalTImer)?

I would not make any Particle API calls from a software timer, especially under 0.5.x. There are better guards in subsequent system versions, but I would assume that no Particle calls like Particle.publish, delay, and Serial (unless you add your own external guard block) are safe to call from a software timer. Delay is not recommended from a software timer because all timers run out of a single thread and if you block the return from a timer it will hold up all other timers.

You definitely cannot make any Particle calls from a hardware timer as it's running in an interrupt context and very few calls are interrupt safe.

m-mcgowan, 2018-11-24:
The most ideal way to help us reproduce and resolve the issue would be to create a small test app that shows the problem.  It would also be a good idea to try on newer versions of firmware.

aeris-ming, 2018-11-24:
@m-mcgowan I have already enable hardware watchdog as a temporary fix.Will functions of system thread be broken by enable watchdog?

aeris-ming, 2019-01-21:
After remove all atomic operation this issue never happen again.It seems because call atomic operation to fast.

closed 2019-01-21 by aeris-ming