particle-iot/device-os #1612: Xenon and Argon as HID over USB device

cbmainz, 2018-11-29:
## Feature Request

I would like to use Xenon and Argon as HID over USB device like it is possible with Photon.

ScruffR, 2019-02-19:
Any ETA?

avtolstoy, 2019-03-01:
At the moment it's definitely not at the top of the list and will require some rework of the USB subsystem on Gen 3 devices in order to introduce both HID and USBSerial1.

jameshallam, 2019-07-19:
I'm also interested in USB HID for the Gen3 devices. Adafruit is currently working on their [Adafruit_TinyUSB_Arduino](https://github.com/adafruit/Adafruit_TinyUSB_Arduino) library, which I've tested on their [Feather nRF52840 Express](https://www.adafruit.com/product/4062), and it seems to work well. Hopefully this might give the Particle team a head start.

NickOveracker, 2019-07-31:
This would be incredibly useful to me as well. I'm currently looking at interface ICs like Microchip's MCP2221, but they are really less than ideal. I'll subscribe to this issue.

DandySoft, 2020-09-25:
I too would like this feature.  Please prioritize.  Your Photon states it is supported but you have been in short supply of those for months.  Would rather implement with Gen3.

andrewsiemer, 2021-02-18:
+1 for HID Support!