particle-iot/device-os #1631: [Argon/Boron] Particle.connect() doesn't call WiFi./Cellular.on()

ScruffR, 2018-12-09:
On Photon/P1/Electron `Particle.connect()` implicitly *also* powered up the radio module when it was off. 
That's currently (0.8.0-rc.25) not happening for the Argon (Boron not tested but probably too)

avtolstoy, 2018-12-10:
I would appreciate if you could provide a minimal example app showcasing the issue. Was `WiFi/Mesh.disconnect()` or `WiFi/Mesh.off()` called as well? Is the device in `AUTOMATIC` or some other mode?

We have multiple network interfaces on Gen 3 devices and in order to simplify the manual management of cloud and network connections the current behavior is for the user application to explicitly enable the necessary network interfaces.

ScruffR, 2018-12-10:
I can do.
`WiFi.off()` was called, but not `Mesh.off()` in `AUTOMATIC` mode.
No `x.disconnect()` was explicitly called.

However
> We have multiple network interfaces on Gen 3 devices and in order to simplify the manual management of cloud and network connections the current behavior is for the user application to explicitly enable the necessary network interfaces.

Since there usually is only one _cloud_ radio (WiFi or Cellular) on each device  and `Particle.connect()` requires the _cloud_ radio to be on, an implicit `on()` call to the one and only radio that relates to it doesn't appear to impair the managability of the connections (same would go for `Mesh` but that's not focus of the issue, but rather the discrepancy between Gen1&2 vs. 3 behaviour in respect of common _cloud_ radio behaviour). 
Needing to call additional functions in order to achive _one_ goal doesn't really seem to be simplifiying things - rather the opposite IMO.
Also, having Gen3 devices behave different than Gen1 & 2 on  intersecting matters seems more confusing that simplifying too IMO. 
Imagine taking a Photon sketch one-to-one to a standalone Argon (e.g. in a Classic Adapter, no mesh required)  as a drop-in replacement and then the previously running code doesn't anymore.

One may argue that a Argon/Boron in an Ethernet shield doesn't need to power on the _cloud_ radio, but that should rather be treated as edge case which should be addressed separately - after all it already has its special treatment in the app and currently concurrency between Ethernet and radio isn't addressed at all AFAICT.

ScruffR, 2019-10-21:
What is the current status of this - the current situation is not really satisfying IMO
e.g. this thread touches on that too
https://community.particle.io/t/xenon-attempting-to-connect-to-cloud-wifi-when-parameters-are-set-for-offline-state/52823

This should suffice (and does with Gen1 & 2)
```
  Particle.connect();  // this alone seems not enough, although it shoule be!
```
But this is required (Gen3 ***only***)
```
  WiFi.on();
  WiFi.connect();  
  if (waitFor(WiFi.ready, 60000)) 
    Particle.connect();
```

markterrill, 2019-11-06:
Yes, this is a major issue. the documentation says that particle.connect wraps wifi activation. that is how it worked since spark core. Post 1.1.1 it doesn't work.

ScruffR, 2020-02-22:
Bump!

Since this previous counter argument is off the table
> We have multiple network interfaces on Gen 3 devices and in order to simplify the manual management of cloud and network connections the current behavior is for the user application to explicitly enable the necessary network interfaces.

Can we please go back to the original behaviour?

_(and remove the mesh tag from this issue)_

avtolstoy, 2020-09-23:
This should be resolved as of DeviceOS 2.x.

closed 2020-09-23 by avtolstoy

chipmc, 2021-01-26:
Is this resolved?  It is still in the docs - https://docs.particle.io/reference/device-os/firmware/boron/#connect--1