particle-iot/device-os #1637: Flasing example app to Argon fails

jizza, 2018-12-14:
## Bug Report
I just received a new Argon. After setting it via the mobile app, I hoped to try out the example app "Blink an LED". Flashing the app however results in a failure.

### Expected Behavior

Flashes app correctly to Argon.

### Observed Behavior

Flasing the app fails (without providing a clear reason)

### Steps to Reproduce

1. Setup via mobile app.
2. Update to firmware 0.8.0-rc.26.
3. Flash "Blink a LED"

### Test App
It's the "Blink an LED" default example. Removed comments for brevity:

```c
int led1 = D0;
int led2 = D7;

void setup() {
  pinMode(led1, OUTPUT);
  pinMode(led2, OUTPUT);
}

void loop() {
  digitalWrite(led1, HIGH);
  digitalWrite(led2, HIGH);

  delay(1000);

  digitalWrite(led1, LOW);
  digitalWrite(led2, LOW);

  delay(1000);
}
```

### Questions
- Does updating to 0.8.0-rc.26 also install Argon NCP to v0.0.5? I tried it via CLI but results in an error.
- How can I get some trace info when flashing? https://console.particle.io/events only says "failed" which isn't very useful.

avtolstoy, 2018-12-14:
Could you provide the output from `particle serial inspect` while the device is in listening mode?

> Does updating to 0.8.0-rc.26 also install Argon NCP to v0.0.5? I tried it via CLI but results in an error.
If you've setup your Argon using the mobile app, you shouldn't need to update the NCP firmware.

If you do want to try updating it, you need to put the device into listening mode, get https://github.com/particle-iot/argon-ncp-firmware/releases/download/v0.0.5/argon-ncp-firmware-0.0.5-ota.bin and flash using `particle flash --serial argon-ncp-firmware-0.0.5-ota.bin`.

You may also try flashing [this](https://drive.google.com/open?id=1ZWNqTYdbJUusZfutC94-rjjAuf1AzSXK) system firmware binary built with `DEBUG_BUILD=y`, manually flashing an application binary with added log handler, triggering a flash from the WebIDE and collecting the log the device produces.

```cpp
// To enable logging on Serial1 @ 115200 bauds
Serial1LogHandler logHandlerSerial1(115200, LOG_LEVEL_ALL);

// To enable logging on Serial (USB)
SerialLogHandler logHandlerSerial(LOG_LEVEL_ALL);
```

vervaekejonathan, 2018-12-17:
I have the same issue, my argon turns magenta for a while and then the led turns off.

avtolstoy, 2018-12-17:
@vervaekejonathan Solid magenta or LED turned off are indications of no chunks coming in from the cloud during OTA. In order to understand the cause, please collect the logs, ideally with the debug `system-part1` I've mentioned in the first reply in this issue.

As for `Error writing firmware: TimeoutError: timed out after 10000ms`, this is a known issue with 0.8.0-rc.26 and current version of particle-cli. It's both an issue in DeviceOS (https://github.com/particle-iot/firmware/pull/1645) and in particle-cli and will be fixed in the next release. The update finishes successfully, despite the fact that the error is being reported.

vervaekejonathan, 2018-12-17:
@avtolstoy do you have an application binary with the log handler enabled? (so I don't have to install the toolchain)

vervaekejonathan, 2018-12-17:
@avtolstoy 

log when uploading the blink led example:

`0000144270 [comm] TRACE: 
0000144270 [comm.coap] TRACE: sending message id=865f
0000144271 [comm] TRACE: update done - missing chunks starting at 0
0000144272 [comm] TRACE: Sent 2 missing chunks
0000144272 [comm.coap] TRACE: sending message id=1e9 synchronously
0000144272 [comm.coap] TRACE: sending message id=1e9
0000144317 [net.nat64] TRACE: NAT64 input (UDP) 52.54.66.84#5684 -> 192.168.0.241#62511
0000144318 [net.nat64] TRACE: No BIB found, trying to create one
0000144318 [net.nat64] TRACE: Not creating a new BIB for a connection initiated from IPv4 side
0000144319 [net.nat64] TRACE: No matching BIB
0000144319 [net.nat64] TRACE: NAT64 input (UDP) 52.54.66.84#5684 -> 192.168.0.241#62511
0000144320 [net.nat64] TRACE: No BIB found, trying to create one
0000144321 [net.nat64] TRACE: Not creating a new BIB for a connection initiated from IPv4 side
0000144321 [net.nat64] TRACE: No matching BIB
0000144472 [net.nat64] TRACE: NAT64 input (UDP) 52.54.66.84#5684 -> 192.168.0.241#62511
0000144472 [net.nat64] TRACE: No BIB found, trying to create one
0000144473 [net.nat64] TRACE: Not creating a new BIB for a connection initiated from IPv4 side
0000144474 [net.nat64] TRACE: No matching BIB
0000144474 [comm.coap] TRACE: recieved ACK for message id=1e9
0000144475 [comm.protocol] INFO: rcv'd message type=6
0000144524 [net.nat64] TRACE: NAT64 input (UDP) 52.54.66.84#5684 -> 192.168.0.241#62511
0000144525 [net.nat64] TRACE: No BIB found, trying to create one
0000144525 [net.nat64] TRACE: Not creating a new BIB for a connection initiated from IPv4 side
0000144526 [net.nat64] TRACE: No matching BIB
0000144527 [net.nat64] TRACE: NAT64 input (UDP) 52.54.66.84#5684 -> 192.168.0.241#62511
0000144527 [net.nat64] TRACE: No BIB found, trying to create one
0000144528 [net.nat64] TRACE: Not creating a new BIB for a connection initiated from IPv4 side
0000144528 [net.nat64] TRACE: No matching BIB
0000144529 [net.nat64] TRACE: NAT64 input (UDP) 52.54.66.84#5684 -> 192.168.0.241#62511
0000144530 [net.nat64] TRACE: No BIB found, trying to create one
0000144530 [net.nat64] TRACE: Not creating a new BIB for a connection initiated from IPv4 side
0000144531 [net.nat64] TRACE: No matching BIB
0000144532 [comm] TRACE: update done received
0000144532 [comm] INFO: Update done 80
0000144533 [comm] TRACE: 
0000144533 [comm.coap] TRACE: sending message id=8662
0000144534 [comm] TRACE: update done - missing chunks starting at 0
0000144534 [comm] TRACE: Sent 2 missing chunks
0000144535 [comm.coap] TRACE: sending message id=1ea synchronously
0000144535 [comm.coap] TRACE: sending message id=1ea
0000144781 [net.nat64] TRACE: NAT64 input (UDP) 52.54.66.84#5684 -> 192.168.0.241#62511
0000144782 [net.nat64] TRACE: No BIB found, trying to create one
0000144782 [net.nat64] TRACE: Not creating a new BIB for a connection initiated from IPv4 side
0000144783 [net.nat64] TRACE: No matching BIB
0000144783 [net.nat64] TRACE: NAT64 input (UDP) 52.54.66.84#5684 -> 192.168.0.241#62511
0000144784 [net.nat64] TRACE: No BIB found, trying to create one
0000144784 [net.nat64] TRACE: Not creating a new BIB for a connection initiated from IPv4 side
0000144785 [net.nat64] TRACE: No matching BIB
0000144786 [comm.coap] TRACE: recieved ACK for message id=1ea
0000144786 [comm.protocol] INFO: rcv'd message type=6
0000144787 [comm] TRACE: update done received
0000144788 [comm] INFO: Update done 80
0000144788 [comm] TRACE: 
0000144788 [comm.coap] TRACE: sending message id=8665
0000144789 [comm] TRACE: update done - missing chunks starting at 0
0000144790 [comm] TRACE: Sent 2 missing chunks
0000144790 [comm.coap] TRACE: sending message id=1eb synchronously
0000144791 [comm.coap] TRACE: sending message id=1eb
0000144836 [net.nat64] TRACE: NAT64 input (UDP) 52.54.66.84#5684 -> 192.168.0.241#62511
0000144836 [net.nat64] TRACE: No BIB found, trying to create one
0000144837 [net.nat64] TRACE: Not creating a new BIB for a connection initiated from IPv4 side
0000144838 [net.nat64] TRACE: No matching BIB
0000144838 [net.nat64] TRACE: NAT64 input (UDP) 52.54.66.84#5684 -> 192.168.0.241#62511
0000144839 [net.nat64] TRACE: No BIB found, trying to create one
0000144840 [net.nat64] TRACE: Not creating a new BIB for a connection initiated from IPv4 side
0000144840 [net.nat64] TRACE: No matching BIB
[... 513 lines cut ...]
0000188004 [ot] INFO: -MLE-----: 	3800 -> 1000, cost:1 1, lqin:3, lqout:3
0000188005 [ot] INFO: -MLE-----: 	5c00 -> 1000, cost:1 1, lqin:3, lqout:3
0000188005 [ot] INFO: -MLE-----: 	9c00 -> fc00, cost:0 16, lqin:0, lqout:0
0000188006 [ot] INFO: -MLE-----: 	dc00 -> 1000, cost:1 1, lqin:3, lqout:3
0000188640 [net.nat64] TRACE: NAT64 input (UDP) 192.168.0.150#33978 -> 192.168.0.255#32412
0000188640 [net.nat64] TRACE: Filtered, broadcast
0000188641 [net.nat64] TRACE: NAT64 input (UDP) 192.168.0.150#52997 -> 192.168.0.255#32414
0000188641 [net.nat64] TRACE: Filtered, broadcast
0000193692 [net.nat64] TRACE: NAT64 input (UDP) 192.168.0.150#33978 -> 192.168.0.255#32412
0000193692 [net.nat64] TRACE: Filtered, broadcast
0000193693 [net.nat64] TRACE: NAT64 input (UDP) 192.168.0.150#52997 -> 192.168.0.255#32414
0000193694 [net.nat64] TRACE: Filtered, broadcast
0000198644 [net.nat64] TRACE: NAT64 input (UDP) 192.168.0.150#33978 -> 192.168.0.255#32412
0000198644 [net.nat64] TRACE: Filtered, broadcast
0000198645 [net.nat64] TRACE: NAT64 input (UDP) 192.168.0.150#52997 -> 192.168.0.255#32414
[... cut ...]
0000203646 [net.nat64] TRACE: NAT64 input (UDP) 192.168.0.150#33978 -> 192.168.0.255#32412
0000203646 [net.nat64] TRACE: Filtered, broadcast
0000203647 [net.nat64] TRACE: NAT64 input (UDP) 192.168.0.150#52997 -> 192.168.0.255#32414
0000203648 [net.nat64] TRACE: Filtered, broadcast
0000203998 [net.nat64] TRACE: NAT64 input (UDP) 192.168.0.150#50702 -> 255.255.255.255#1900
0000203998 [net.nat64] TRACE: Filtered, broadcast
0000204049 [net.nat64] TRACE: NAT64 input (UDP) 192.168.0.150#50702 -> 255.255.255.255#1900
0000204049 [net.nat64] TRACE: Filtered, broadcast
`

jizza, 2018-12-17:
I’m sorry I can’t reply to this thread right now. I’m struggling with another issue which prevents me from even claiming a device.
https://community.particle.io/t/recover-from-cli-command-particle-keys-server/46274

vervaekejonathan, 2018-12-18:
@avtolstoy 
Argon with your debug build:

> particle flash --usb Downloads/system-part1-0.8.0-rc.26-debug-argon.bin

start serial monitor via
> particle usb monitor

Upload default blink-a-led.ino example via web IDE. this is the log:

https://shrib.com/#dInO3MUNITHJTB--hubG

Console log:
[image]

avtolstoy, 2018-12-19:
@vervaekejonathan Thank you for the log. I see that the packets are coming in (presumably with chunks) but for some reason they are not being processed by the communication layer. Please collect the log with the following [system-part1](https://drive.google.com/open?id=1-GdtCH2RBwZDcA7CXZfehv3yAmGn2YG8), which enables debug logging in our TCP/IP stack, that should give the full picture.

>  Could you tell us more about your environment?
I would also still like to hear some information on what WiFi router you are using etc.

jizza, 2018-12-23:
> particle serial inspect

@avtolstoy 

```
particle serial inspect
Platform: 12
Modules
  Bootloader module #0 - version 215, main location, 49152 bytes max size
    Integrity: PASS
    Address Range: PASS
    Platform: PASS
    Dependencies: PASS
  System module #1 - version 325, main location, 671744 bytes max size
    Integrity: PASS
    Address Range: PASS
    Platform: PASS
    Dependencies: PASS
      Bootloader module #0 - version 101
      undefined module #0 - version 0
  User module #1 - version 5, main location, 131072 bytes max size
    UUID: 72B73BFAB5C902DBE9EC90A13958C90408C6FA4758B9F9C5CFB5100141C1C58B
    Integrity: PASS
    Address Range: PASS
    Platform: PASS
    Dependencies: **FAIL**
      System module #1 - version 326
      undefined module #0 - version 0
  empty - factory location, 131072 bytes max size
  empty - main location, 1536000 bytes max size
```

How can i resolve the failed dependencies on User module #1?

closed 2019-01-06 by jizza