particle-iot/device-os #1641: [Mesh] Possible issue with delayMicroseconds() causes libraries to fail

ScruffR, 2018-12-15:
As mentioned by a member in this post, there might be some jitter in the way how `delayMicroseconds()` is implemented.
https://community.particle.io/t/argon-dht11-nothing-but-zeros/46108/24?u=scruffr

With libraries that rely on the accuracy of `delayMicroseconds()` (e.g. Adafruit_DHT`) this might be a breaking issue.

luiscamaral, 2018-12-15:
Hi ScruffR, thanks for filing this issue.

I've used a very simple sketch for this test running on a Xenon board. Sketch was compiled using workbench local compiler and flash thru serial. 

I am using a FTserial to monitor Serial 1 and DSLogic digital analyzer to identify and decode protocol from the pin D11.

From the digital analyzer we can see clearly that the first 20ms delay using delay(20) function from Adafruit library is working fine. But the delayMicroseconds(40) is showing only 12.25us.

The same code is running well on photon board.

Bellow the Adafruit Library fragment for reference. [Adafruit_DHT.cpp](https://build.particle.io/libs/Adafruit_DHT/0.0.2/tab/Adafruit_DHT.cpp)

```
...
// pull the pin high and wait 250 milliseconds
	digitalWrite(_pin, HIGH);
	delay(250);

// now pull it low for ~20 milliseconds
	pinMode(_pin, OUTPUT);
	digitalWrite(_pin, LOW);
	delay(20);
	noInterrupts();
	digitalWrite(_pin, HIGH);
	delayMicroseconds(40);
	pinMode(_pin, INPUT);
...
```

--

YutingYou, 2018-12-20:
Hi @ScruffR  @luiscamaral ,
`delayMicroseconds()` works fine, The MCU clock on Mesh Device is much slower than Photon/Electron (64MHz vs 120MHz). This issue is caused by performance lost in `digitalWrite()`, there is a safety check inside it.

Below are the tests for `digitalWrite()` and `fastPin()`:
```
SYSTEM_MODE(MANUAL);

void setup() {
    pinMode(D2, OUTPUT);
}

void loop() {
    while(1) {
        delayMicroseconds(40);
        digitalWrite(D2, HIGH);
        delayMicroseconds(40);
        digitalWrite(D2, LOW);
    }
}
```
[image]

```
void setup() {
    pinMode(D2, OUTPUT);
}

void loop() {
    while(1) {
        delayMicroseconds(40);
        pinSetFast(D2);
        delayMicroseconds(40);
        pinResetFast(D2);
    }
}
```
[image]

avtolstoy, 2019-02-28:
@ScruffR Can we consider this issue resolved?

ScruffR, 2019-02-28:
I think we can and I guess the recent timing improvements done in 0.9.0 help too.

However in connection with this issue: I had made out with @YutingYou and @monkbroc that it would be a good idea when Particle took over the `Adafruit_DHT` library as the original contributor seems to have abandoned this library and above fix should be applied but nothing in that regard seems to have happened since then.
I'll also refresh the Slack thread on that.

technobly, 2019-03-11:
The library has been updated to support Mesh (Adafruit_DHT v0.0.3)
https://build.particle.io/libs/Adafruit_DHT/0.0.3/tab/Adafruit_DHT.cpp

Please close this issue if it is resolved, thanks!

ScruffR, 2019-03-11:
The library is missing the GitHub repository link tho'
@YutingYou, can you add that and reupload the library?

When that's done I'll close this issue.

YutingYou, 2019-03-12:
@ScruffR, I've created the sensor library in my GitHub and re-upload it, please close the issue. Thank you!

ScruffR, 2019-03-12:
Thanks for that :+1:

closed 2019-03-12 by ScruffR