particle-iot/device-os #1654: [Mesh rc.27] Regression causing PietteTech_DHT library to fail

ScruffR, 2018-12-22:
As discussed in this thready
https://community.particle.io/t/argon-dht11-nothing-but-zeros/46108/31?u=scruffr

While the library sample [`dht_simple.ino`](https://github.com/eliteio/PietteTech_DHT/tree/master/examples/DHT_simple) works perfectly fine with rc.26 for DHT11 & DHT22 sensors, rc.27 has introduced major problems for that library that worked flawless on Gen1&2 for years.
With a DHT22 I get max 3 to 4 good readings followed by a couple of errors and with a DHT11 no good readings at all.
Since this library is interrupt based and there were some "fixes" in connection with interrupts I'm inclined some of these fixes have actually unfixed something else.

Tested on Argon 0.8.0-rc.26 & -rc.27

YutingYou, 2019-01-03:
Hi @ScruffR, I can't reproduce this issue on Argon-RC27 modular firmware with DHT11. I used manual mode.

```
Sat Jan  1 00:00:32 2000
Humid: 46.00% - Temp: 23.00*C 73.40*F 296.15*K - DewP: 10.75*C - HeatI: 25.11*C
Sat Jan  1 00:00:39 2000
Humid: 49.00% - Temp: 23.00*C 73.40*F 296.15*K - DewP: 11.70*C - HeatI: 25.09*C
Sat Jan  1 00:00:41 2000
Humid: 61.00% - Temp: 23.00*C 73.40*F 296.15*K - DewP: 15.07*C - HeatI: 24.71*C
Sat Jan  1 00:00:43 2000
Humid: 74.00% - Temp: 23.00*C 73.40*F 296.15*K - DewP: 18.11*C - HeatI: 23.76*C
Sat Jan  1 00:00:46 2000
Humid: 83.00% - Temp: 23.00*C 73.40*F 296.15*K - DewP: 19.95*C - HeatI: 22.78*C
Sat Jan  1 00:00:48 2000
Humid: 89.00% - Temp: 23.00*C 73.40*F 296.15*K - DewP: 21.08*C - HeatI: 21.98*C
Sat Jan  1 00:00:50 2000
Humid: 93.00% - Temp: 24.00*C 75.20*F 297.15*K - DewP: 22.79*C - HeatI: 23.32*C
Sat Jan  1 00:00:52 2000
Humid: 95.00% - Temp: 24.00*C 75.20*F 297.15*K - DewP: 23.15*C - HeatI: 23.12*C
Sat Jan  1 00:00:55 2000
Humid: 94.00% - Temp: 24.00*C 75.20*F 297.15*K - DewP: 22.97*C - HeatI: 23.22*C
Sat Jan  1 00:00:57 2000
Failed to read from DHT sensor!
Humid: 91.00% - Temp: 25.00*C 77.00*F 298.15*K - DewP: 23.42*C - HeatI: 25.63*C
Sat Jan  1 00:01:02 2000
Humid: 89.00% - Temp: 25.00*C 77.00*F 298.15*K - DewP: 23.06*C - HeatI: 25.68*C
Sat Jan  1 00:01:04 2000
Humid: 88.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 23.85*C - HeatI: 27.96*C
Sat Jan  1 00:01:06 2000
Humid: 86.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 23.47*C - HeatI: 27.89*C
Sat Jan  1 00:01:08 2000
Humid: 84.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 23.08*C - HeatI: 27.81*C
Sat Jan  1 00:01:11 2000
Humid: 82.00% - Temp: 27.00*C 80.60*F 300.15*K - DewP: 23.66*C - HeatI: 29.93*C
Sat Jan  1 00:01:13 2000
```

ScruffR, 2019-01-03:
Can you try AUTOMATIC mode?
I'm not sure how I tried, but since the linked sample is *not* using MANUAL mode, I'm pretty sure I wasn't either. 
However, the library did and is supposed to run with AUTOMATIC mode too.

YutingYou, 2019-01-04:
I saw the feedback from @syrinxtech that the driver worked for him after removing manual mode. I also tried AUTOMATIC mode, it worked fine.

My test code:
https://go.particle.io/shared_apps/5c2ec562e71c122067000cd4

```
Fri Jan  4 02:34:11 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:14 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:16 2019
Failed to read from DHT sensor!
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:20 2019
Failed to read from DHT sensor!
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:25 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:27 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:30 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:32 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:34 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:36 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:39 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:41 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:43 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:46 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:48 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:50 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:52 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:55 2019
Humid: 51.00% - Temp: 26.00*C 78.80*F 299.15*K - DewP: 15.07*C - HeatI: 26.60*C
Fri Jan  4 02:34:57 2019
Failed to read from DHT sensor!
Humid: 50.00% - Temp: 25.00*C 77.00*F 298.15*K - DewP: 13.84*C - HeatI: 25.89*C
Fri Jan  4 02:35:01 2019
Humid: 51.00% - Temp: 25.00*C 77.00*F 298.15*K - DewP: 14.15*C - HeatI: 25.90*C
Fri Jan  4 02:35:04 2019
Humid: 51.00% - Temp: 25.00*C 77.00*F 298.15*K - DewP: 14.15*C - HeatI: 25.90*C
Fri Jan  4 02:35:06 2019
Humid: 51.00% - Temp: 25.00*C 77.00*F 298.15*K - DewP: 14.15*C - HeatI: 25.90*C
Fri Jan  4 02:35:08 2019
Humid: 51.00% - Temp: 25.00*C 77.00*F 298.15*K - DewP: 14.15*C - HeatI: 25.90*C
Fri Jan  4 02:35:11 2019
Failed to read from DHT sensor!
Humid: 51.00% - Temp: 25.00*C 77.00*F 298.15*K - DewP: 14.15*C - HeatI: 25.90*C
Fri Jan  4 02:35:15 2019
```

ScruffR, 2019-01-04:
You seem to ***not*** compare the correct things.
The title of this issue refers to the PietteTech_DHT library but in your tests you use the Adafruit_DHT library. That won't help pinpointing the issue.

The issue revolves (most likely) around the change in behaviour in regards to *interrupts* and the PietteTech_DHT library uses *interrupts* hence it is (was) the prefered library for most people.

YutingYou, 2019-01-04:
Thanks for pointing it out, I misunderstood it.

ScruffR, 2019-01-22:
Any update on this front?

YutingYou, 2019-01-24:
Hi @ScruffR , sorry to reply you so late.

This issue is caused by the error of `millis()` and `micros()`, we changed the clock source from `DWT->CYCCNT` to RTC2  to resolve OT stability issues in rc.27, so the time error become : 1 / 32.768KHz = 30us.

Simply you can modify the error range in the driver if you want to run it in RC27
```
// if (125 < delta && delta < 200)
 if (125 < delta && delta < 220)
```
Application source: https://go.particle.io/shared_apps/5c496d003c3eb32afb0002ba

Or you can wait for next release.

ScruffR, 2019-01-26:
Thanks :+1:
I've updated the library to v0.0.6 which implements your change

Do you expect the fix in next release would break that correction or can we leave it at that then?

YutingYou, 2019-01-28:
This fix should be compatible in the next release.

ScruffR, 2019-01-28:
Thanks, guess we can close this then :+1:

ScruffR, 2019-01-28:
I'll wait with closing this for good for the next release and test 0.0.5 again to see whether the "fix" of underlying issue actually works.

avtolstoy, 2019-02-28:
@ScruffR Could you please confirm whether #1682 included in 0.9.0 resolves this issue?

ScruffR, 2019-02-28:
Will do, have to try with an older version of the library (find out which first)

ScruffR, 2019-03-01:
@avtolstoy, I can say 0.9.0 did improve things with the library version 0.0.5 (prior the workaround) but it's still not as reliable as it was on Gen1&2.
Gen3 reanders avg. 9 good readings between errors Gen1&2 ~50.

gstrodel, 2019-04-03:
I am still seeing this error on 3 different Xenons all running 0.9.0. and the same sketch. My code: https://go.particle.io/shared_apps/5ca40ba014adde00161bedce