particle-iot/device-os #1668: Arduino library compatibility needs #define SPI_HAS_TRANSACTION 1

ScruffR, 2019-01-08:
Many recent Arduino libraries rely on the existence of `#define SPI_HAS_TRANSACTION 1` when the platform supports SPI transactions.
Since Particle devices do, we should also have that macro defined.

e.g. these libraries
https://github.com/technologiescollege/arduino/blob/master/hardware/arduino/avr/libraries/SPI/src/SPI.h
https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_SPITFT_Macros.h
https://github.com/adafruit/Adafruit_VS1053_Library/blob/master/Adafruit_VS1053.cpp
https://github.com/adafruit/Adafruit_SSD1305_Library/blob/master/Adafruit_SSD1305.cpp
https://github.com/esp8266/Basic/blob/master/libraries/Adafruit_STMPE610/Adafruit_STMPE610.cpp

ScruffR, 2019-01-12:
This also touches on a realated issue
https://github.com/particle-iot/cloud/issues/57

ScruffR, 2019-07-14:
Still no comment?
Another member got bitten by that https://community.particle.io/t/ssd1305-issues-on-spi/50958

psherk-cryo, 2019-07-15:
Yep, the lack of this definition cost us a lot of time, specifically related to SDFat and an SSD1305 OLED driver.  We built our own library for the OLED, but it looks like SDFat uses the SPI_HAS_TRANSACTION define, which caused a lot of very random and hard to debug issues.  Please fix this one Particle.

karlg100, 2020-01-01:
I'll bump this issue.  many Adafruit libraries seem to be nonfunctional from this.

pkourany, 2020-02-12:
Bumping again.

PR #2023 (2020-02-13): [wiring] Fixes a regression in SPI due to thread-safety additions from #1879
### Problem

1. Using SPI object in global object constructors leads to a crash due to undefined initialization order of global objects
2. `pinAvailable()` calls `SPI.isEnabled()`, potentially from interrupt context.
3. #1668

### Solution

1. Use singleton-ish type of initialization for SPI objects with a function returning a reference to static object
2. Remove locking from `SPI.isEnabled()`, since in HAL it is anyway implemented with a volatile variable.
3. Adds `SPI_HAS_TRANSACTION` define

### Steps to Test

N/A
 [... cut]
closed 2020-02-13 by avtolstoy