particle-iot/device-os #1671: [Mesh] SPI.begin() misbehaves by

ScruffR, 2019-01-12:
On Gen3/Mesh devices it's currently virtually impossible to have two libraries using SPI work together without extra efforts.
Due to the fact that the mesh implementation of `SPI.begin(pin)` resets any previously set CS pin from OUTPUT to Hi-Z/INPUT when calling `SPI.begin(pin)` anew for another slave with its own CS pin. This prevents the library that had assigned it's CS pin before from useing `digitalWrite(pin, LOW)` in order to activate the slave.

A simple test sketch shows the difference between Gen2 and Gen3 devices in this respect
```
void setup() {
    Particle.function("SPITest", spiTest);
}

int spiTest(const char* pin) {
    int p = atoi(pin);
    if (p > 0) {
        SPI.begin(p);
    }
    return p;
}
```
When first calling `SPITest` for pin 7 (LED pin) the LED goes on. With a subsequent call for any other pin on e.g. an Argon the LED goes out indicating that the CS pin is not driven HIGH anymore, but on e.g. Photon the LED stays on.

This may also relate to this issue
https://github.com/particle-iot/device-os/issues/1290

---

A discussion about this has already been lead with @YutingYou on Slack (Thread started 01/11/2019 9:23pm UTC)

avtolstoy, 2019-03-01:
@ScruffR Please confirm that this issue was resolved in 0.9.0 by #1672. Thanks!

ScruffR, 2019-03-01:
Yes, this has been resolved with 0.9.0.

closed 2019-03-01 by ScruffR