particle-iot/device-os #1678: FuelGauge declaration inside loop causes Boron to malfunction

tompropst, 2019-01-22:
## Bug Report

### Expected Behavior

Expect to use a local `FuelGauge` object in a function such as `void loop()`.

### Observed Behavior

Declaring a local `FuelGauge` object in `void loop()` causes Boron to stop working. Any mesh devices will no longer connect. You must enter safe mode and remove the local `FuelGauge` object to restore any functionality.

### Steps to Reproduce

Using tinker-0.8.0-rc.27-boron.bin, flash the test app below to a Boron device.

### Test App

```c
double batt_v = 0.0;

void setup() {
  Particle.variable("batt_v", batt_v);
}

void loop() {
  FuelGauge fuel;
  batt_v = fuel.getVCell();
  delay(1000);
}
```

### References

The [FuelGauge documentation](https://docs.particle.io/reference/device-os/firmware/boron/#fuelgauge) is not clear on whether the `FuelGauge` object must be global.

---

daniel-HooperEngineering, 2019-03-24:
It seems as though FuelGauge was intended to work as an Arduino-style singleton object, like Serial, Serial1, Wire etc, the only difference being that the object is not pre-instantiated like those other examples (leaving it up to you whether to do so).
I would suggest that this is a documentation issue rather than a firmware bug.

technobly, 2019-12-05:
@tompropst @daniel214 thanks for the bug report!  Sorry for the late reply :( I don't know when this was fixed but appears to be resolved as of 1.4.2 at least.  Could you try again with an updated version of Device OS?

technobly, 2019-12-12:
@tompropst @daniel214 I'm going to close this issue due to it being unreproducible (looks resolved).  Please reopen if you disagree.  Thanks!

closed 2019-12-12 by technobly