particle-iot/device-os #1679: Particle.publish Future appears to complete before send is finalized

Richard-Mansfield-4, 2019-01-22:
Greetings :)

I have been attempting to trim down the time our device has to be connected to the network (Photon on Wifi), and the confirmation check for Particle.publish(,,,WITH_ACK) doesn't appear to be functioning properly.

If I check the Future result and then immediately sleep, the message is never received by the cloud. If I merely insert a delay after the send, the message is received. I would expect that when success is set, the message has been actually sent (or not if it failed), and if the Future returns true the message appears in the cloud and it is safe to disconnect. This is not the case.

Additional debug output shows that the publish future returns true even though the message is not yet delivered, and in fact never appears in the cloud.

Example (firmware 1.0.0). If the delay line is uncommented, "bar" appears in the cloud. If the delay is not present, "bar" never appears.
```

SYSTEM_MODE(MANUAL);

void setup() {
  Serial.begin(9600);
}

int x = 0;

void loop() {
  
    Particle.connect();
    Particle.process(); 

    particle::Future<bool> successfuture = Particle.publish("foo", "bar", PRIVATE, WITH_ACK);
    bool success = successfuture;
    if (success == true)
        {++x;}

    Serial.write(x); //Make sure the compiler actually has to test success. This has no effect.
    __sync_synchronize(); //Make sure the above really did execute. This has no effect.

    //delay(1000000); //If uncommented, this causes the message to be received by the cloud. If this line is commented, the message does not appear.
    System.sleep(SLEEP_MODE_DEEP, 5);
}

```

WellBotLLC, 2019-11-30:
Seeing a similar issue. However, I am getting false negatives. The publish is successful, but the ACK is returning false. Running 1.2.1 on Electron. SYSTEM_THREAD(ENABLED), SYSTEM_MODE(SEMI_AUTOMATIC).

zfields, 2019-12-18:
I extended the testing of the `particle::Future` and it looks like the `publish()` is erroring out.

```c++
...
    particle::Future<bool> successfuture = Particle.publish("foo", "bar", PRIVATE, WITH_ACK);
    //bool success = successfuture;
    bool success = successfuture.wait();
    if (success == true) {
      if (successfuture.isCancelled()) {
        ++x;
      } else if (successfuture.isFailed()) {
        auto err = successfuture.error();
      } else if (successfuture.isSucceeded()) {
        x = 0;
      }

      Serial.write(x); //Make sure the compiler actually has to test success. This has no effect.
...
```

Looking at this line `auto err = successfuture.error();` under the debugger showed

```
err: {...}
- private
-- msg_: 0x0
--- *msg_: 0 '0\000'
-- type_: particle::Error::INVALID_STATE
```

zfields, 2019-12-18:
Okay, I've got it working on a Photon running `v1.4.4`

The key was continue processing, until `Particle.connected()` returns `true`.

```c++
    for(;!Particle.connected();) {
      Particle.process();
    }
```

`successfuture` is a bit of a misnomer, because it really only indicates that the thread has finished working; regardless of whether it succeeded, failed or was cancelled.

When inspecting the error under the debugger, we saw `particle::Error::INVALID_STATE`, which is returned when the device is not "connected". `Particle.process()` allows the background threads that maintenance the state of the connection to do their work. So if we call `Particle.process()` while `Particle.connected()` is not `true`, then we give the system an opportunity to get itself into a connected state.

Richard-Mansfield-4, 2019-12-18:
Yeah. With 1.4+, it seems to work as expected now. And yes, my example code above was broken. (Not being connected wasn't the issue originally.) Sorry about that. The actual problem seems to have been fixed at some point.

closed 2019-12-18 by Richard-Mansfield-4