particle-iot/device-os #1701: SOS when removing USB/Serial from battery powered device

GregSawin, 2019-02-27:
## Bug Report
A battery powered Xenon or Argon with Serial open will crash if you unplug the usb

Upon restart I see this event: spark/device/last_reset - panic, assert_failed</details>

From ScruffR:
I can reproduce that with my 0.9.0 Xenons and Argon too (Boron not yet tested).  In this case for example nothing happens when there is no USB CDC connection open at the time of disconnect but when someone is listening (e.g. particle serial monitor) then the SOS+10 panic occurs.
### Expected Behavior

Not crash

### Observed Behavior

Crash

### Steps to Reproduce

Setup an Argon/Xenon with a Lipo Battery.  Flash the test app code with firmware 0.9.0, connect to the Serial connection via USB, then unplug the USB and you'll see the SOS.

### Test App

```c
void setup() {
    Serial.begin();
}

void loop() {
    Serial.print("millis: ");
    Serial.println(millis());
    delay(1000);
}
```

### References

https://community.particle.io/t/sos-when-removing-usb-serial-from-battery-powered-xenon/48076

---

avtolstoy, 2019-02-28:
Stack trace:
```
#0  panic_ (code=code@entry=AssertionFailure, extraInfo=extraInfo@entry=0x0, HAL_Delay_Microseconds=0x4caad <HAL_Delay_Microseconds(uint32_t)>) at src/panic.c:41
#1  0x00045f0c in usb_uart_send (data=data@entry=0x20016faf "m\270S", size=size@entry=1) at src/nRF52840/usb_hal_cdc.c:378
#2  0x0004ce7e in HAL_USB_USART_Send_Data (serial=<optimized out>, data=109 'm') at src/nRF52840/usb_hal.cpp:91
#3  0x000d476e in ?? ()
```

@YutingYou Please take a look at this issue. Thanks.

avtolstoy, 2020-09-23:
This should be resolved at least as of DeviceOS 2.0.0-rc.2 with https://github.com/particle-iot/device-os/pull/2101

closed 2020-09-23 by avtolstoy