particle-iot/device-os #1702: [Mesh 0.9.0] Assertion failure SOS when unplugging device with open USB Serial connection

ScruffR, 2019-02-27:
While battery driven Gen1&2 devices can happily cope with being unplugged from the USB host while a USB CDC connection is open Gen3 devices crash with an SOS+10 Assertion Failure panic.
Unplugging while there is no serial terminal open does _not_ lead to a crash.

Tested with Argon and Xenon running 0.9.0 on a Win10 machine and `particle serial monitor --follow`

Test code
```
void setup() {
    Serial.begin();
}

void loop() {
    Serial.print("millis: ");
    Serial.println(millis());
    delay(1000);
}
```
Discussion here
https://community.particle.io/t/sos-when-removing-usb-serial-from-battery-powered-xenon/48076

avtolstoy, 2019-02-28:
This seems to be a duplicate of #1701, let's close one or the other.

ScruffR, 2019-02-28:
Must have filed it about the same time

closed 2019-02-28 by ScruffR