particle-iot/device-os #1704: Particle Mesh Deep Sleep hardcoded to D8

steves2j, 2019-02-27:
<details>
  <summary><i>Deep Sleep on the Argon/Boron/Xenon hardcoded to D8 when it doesn't need to be.</i></summary>

</details>

## Feature/Enhancement Request

In version 0.9.0 with the introduction of sleep functionality, the deep sleep has been hardcoded to D8 to mirror the Gen2 functionality where in Gen3 you can actually use different pins as a WKP pin.

Removing the hardcoded-ness of this would allow a developer to use a different Wake up pin from the hardcoded one.

Feature Request that a System.enableFeature or the ability to pass in a PIN ID to the core_hal.c  for mesh products would be useful and make the product more flexible.

### Test App

Testing alternative pins using the following from User space demonstrates the change of pin from D8 to A0.

```
    uint8_t st = 0;
    sd_nvic_critical_region_enter(&st);
    nrf_lpcomp_disable();
    HAL_Interrupts_Suspend();
    NRF5x_Pin_Info* PIN_MAP = HAL_Pin_Map();
    uint32_t nrf_pin = NRF_GPIO_PIN_MAP(PIN_MAP[A0].gpio_port, PIN_MAP[A0].gpio_pin);
    nrf_gpio_cfg_sense_input(nrf_pin, NRF_GPIO_PIN_PULLDOWN, NRF_GPIO_PIN_SENSE_HIGH);
    nrf_pwm_disable(NRF_PWM0);
    nrf_pwm_disable(NRF_PWM1);
    nrf_pwm_disable(NRF_PWM2);
    nrf_pwm_disable(NRF_PWM3);
    sd_power_system_off();
    while (1);
```

avtolstoy, 2019-02-28:
Thank you for the feature request! At the moment we are targeting feature parity with Gen 2 devices, but it is on our roadmap to introduce more flexibility to sleep modes on Gen 3 devices.

pkourany, 2019-05-03:
@avtolstoy, as we now know, feature parity is not really possible since the nRF52840 use a _level_ interrupt instead of an edge interrupt.  Given that Gen3 breaks things, we might as well break it completely!

avtolstoy, 2020-02-06:
Configurable with Sleep 2.0 (#1986)

closed 2020-02-06 by avtolstoy