particle-iot/device-os #1716: Particle.variable() fails on type int

jayhale, 2019-03-07:
## Bug Report

- OS 0.9.0
- Argon

### Expected Behavior

Register a variable (type `int`) with Particle cloud.

### Observed Behavior

Fails to compile, error originating from the `Particle.variable()` statement:

```
In file included from ../wiring/inc/spark_wiring.h:51:0,
                 from ./inc/application.h:40,
                 from simple-variable.cpp:1:
../wiring/inc/spark_wiring_cloud.h: In instantiation of 'static bool CloudClass::variable(const T&, const Types& ...) [with T = int; Types = {int}]':
simple-variable.ino:4:49:   required from here
../wiring/inc/spark_wiring_cloud.h:85:25: error: no matching function for call to 'CloudClass::_variable(const int&, const int&)'
         return _variable(name, args...);
```
*Full output below*

### Steps to Reproduce

Attempt to compile the test app below.

### Test App

```c
int accumulator = 0;

void setup() {
    Particle.variable('accumulator', accumulator);
}

void loop() {
    accumulator++;
    delay(1000);
}
```

### References
Documentation: https://docs.particle.io/reference/device-os/firmware/photon/#particle-variable-

### Full build output
```
Processing  simple-variable.ino
make -C ../modules/argon/user-part all
make[1]: Entering directory '/firmware/modules/argon/user-part'

make -f ../../../modules/shared/nRF52840/build_linker_script.mk PREBUILD=1
make[2]: Entering directory '/firmware/modules/argon/user-part'
Creating module_user_memory.ld ...
make[2]: Leaving directory '/firmware/modules/argon/user-part'

make -C ../../../user 
make[2]: Entering directory '/firmware/user'
Building cpp file: simple-variable.cpp
Invoking: ARM GCC CPP Compiler
mkdir -p ../build/target/user/platform-12-m
arm-none-eabi-gcc -DSTM32_DEVICE -DnRF52840 -DNRF52840_XXAA -DPLATFORM_THREADING=1 -DPLATFORM_ID=12 -DPLATFORM_NAME=argon -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD00C -DUSBD_PID_CDC=0xC00C -DSPARK_PLATFORM -g3 -gdwarf-2 -Os -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DINCLUDE_PLATFORM=1 -DSOFTDEVICE_PRESENT=1 -DS140 -DPRODUCT_ID=12 -DPRODUCT_FIRMWARE_VERSION=65535 -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE=\"openthread-config-project.h\" -DENABLE_FEM=1 -DNRF_802154_PROJECT_CONFIG=\"openthread-platform-config.h\" -DRAAL_SOFTDEVICE=1 -D_WIZCHIP_=W5500 -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -DLFS_CONFIG=lfs_config.h -DSYSTEM_VERSION_STRING=0.9.0 -DRELEASE_BUILD -I./inc -I../wiring/inc -I../system/inc -I../third_party/miniz/miniz -I../services/inc -I../third_party/nanopb/nanopb -I../communication/src -I../hal/inc -I../hal/shared -I../hal/src/argon -I../hal/network/lwip/esp32 -I../hal/src/nRF52840 -I../hal/src/armv7 -I../hal/src/nRF52840 -I../hal/src/nRF52840/lwip -I../hal/src/nRF52840/freertos -I../hal/src/nRF52840/openthread -I../hal/src/nRF52840/mbedtls -I../hal/src/nRF52840/littlefs -I../hal/network/api -I../hal/network/lwip -I../hal/network/lwip/posix -I../hal/network/openthread -I../hal/network/lwip/wiznet -I../hal/network/ncp -I../hal/network/ncp/at_parser -I../third_party/lwip/lwip/src/include -I../third_party/freertos/freertos/FreeRTOS/Source/include -I../third_party/nrf5_sdk/nrf5_sdk/external/freertos/portable/GCC/nrf52 -I../third_party/nrf5_sdk/nrf5_sdk/external/freertos/portable/CMSIS/nrf52 -I../third_party/openthread/openthread/include -I../third_party/openthread/openthread/src/core -I../third_party/openthread/openthread/examples/platforms -I../third_party/openthread/openthread/third_party/NordicSemiconductor/drivers/radio -I../third_party/openthread/openthread/third_party/NordicSemiconductor/drivers/radio/hal -I../third_party/openthread/openthread/third_party/NordicSemiconductor/drivers/radio/rsch -I../third_party/openthread/openthread/third_party/NordicSemiconductor/drivers/radio/rsch/raal -I../third_party/openthread/openthread/third_party/NordicSemiconductor/drivers/radio/rsch/raal/softdevice -I../third_party/wiznet_driver/wiznet_driver/Ethernet -I../gsm0710muxer/gsm0710muxer/include -I../platform/shared/inc -I../third_party/nrf5_sdk -I../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers/include -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/hal -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/mdk -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/hal -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers/src/prs -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/util -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers/nrf52 -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/common -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/fstorage -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/experimental_section_vars -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/atomic -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/crc32 -I../third_party/nrf5_sdk/nrf5_sdk/components/drivers_nrf/usbd -I../third_party/nrf5_sdk/nrf5_sdk/integration/nrfx/legacy -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/delay -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/experimental_log -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/experimental_log -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/experimental_log/src -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/experimental_section_vars -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/atomic -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/delay -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/queue -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/common -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/mbr/nrf52840/headers -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140 -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers/nrf52 -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/fifo -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/atomic_fifo -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/strerror -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/fstorage -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/atomic_flags -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/usbd -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/usbd/class/cdc -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/usbd/class/cdc/acm -I../third_party/nrf5_sdk/nrf5_sdk/components/ble/common -I../third_party/nrf5_sdk/nrf5_sdk/components/ble/nrf_ble_gatt -I../third_party/nrf5_sdk/nrf5_sdk/components/ble/ble_advertising -I../third_party/nrf5_sdk/nrf5_sdk/external/nrf_cc310/include -I../third_party/littlefs/littlefs -I../platform/MCU/nRF52840/inc -I../dynalib/inc -I -I./libraries -I. -MD -MP -MF ../build/target/user/platform-12-msimple-variable.o.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -DSPARK_PLATFORM_NET=ESP32 -fno-builtin-malloc -fno-builtin-free -fno-builtin-realloc  -DLOG_INCLUDE_SOURCE_INFO=1 -DPARTICLE_USER_MODULE -DUSER_FIRMWARE_IMAGE_SIZE=0x20000 -DUSER_FIRMWARE_IMAGE_LOCATION=0xD4000 -DMODULAR_FIRMWARE=1 -DMODULE_VERSION=5 -DMODULE_FUNCTION=5 -DMODULE_INDEX=1 -DMODULE_DEPENDENCY=4,1,403 -DMODULE_DEPENDENCY2=0,0,0 -D_WINSOCK_H -D_GNU_SOURCE -DLOG_MODULE_CATEGORY="\"app\""  -fno-exceptions -fno-rtti -fcheck-new -std=gnu++11 -c -o ../build/target/user/platform-12-msimple-variable.o simple-variable.cpp
simple-variable.ino:4:23: warning: character constant too long for its type
     Particle.variable('accumulator', accumulator);
                       ^
In file included from ../wiring/inc/spark_wiring.h:51:0,
                 from ./inc/application.h:40,
                 from simple-variable.cpp:1:
../wiring/inc/spark_wiring_cloud.h: In instantiation of 'static bool CloudClass::variable(const T&, const Types& ...) [with T = int; Types = {int}]':
[... 98 lines cut ...]
         return _variable(name, args...);
                         ^
../wiring/inc/spark_wiring_cloud.h:173:24: note: candidate: template<class T> static bool CloudClass::_variable(const T*, const String*, const CloudVariableTypeString&)
[... cut ...]
                         ^
../build/module.mk:277: recipe for target '../build/target/user/platform-12-msimple-variable.o' failed
make[2]: Leaving directory '/firmware/user'
make[2]: *** [../build/target/user/platform-12-msimple-variable.o] Error 1
../../../build/recurse.mk:11: recipe for target 'user' failed
make[1]: Leaving directory '/firmware/modules/argon/user-part'
make[1]: *** [user] Error 2
../build/recurse.mk:11: recipe for target 'modules/argon/user-part' failed
make: *** [modules/argon/user-part] Error 2
```

avtolstoy, 2019-03-07:
You are using single quotes in `Particle.variable('accumulator', accumulator);` instead of double quotes. The line should be `Particle.variable("accumulator", accumulator);`

jayhale, 2019-03-07:
Dang - should have caught that - thanks!

closed 2019-03-07 by jayhale