particle-iot/device-os #1718: I2C stuck in loop after sleep on Argon.

steves2j, 2019-03-11:
If the particle is put to sleep and wakes up while in the mid of doing a I2C transfer, the resuming will leave the particle in a stuck state.

Attaching a debugger leads me to believe that the i2c hal is stuck in a loop.

Looking at hal\src\nRF52840\i2c_hal.cpp around line 341 there is a while loop that has no timeout or getout clause if ever the transfer_state is stuck at TRANSFET_STAT_BUSY.

YutingYou, 2019-03-13:
Hi @steves2j, thanks for reporting, I can't reproduce the issue with my sensor MPU9250, can you please share your code?

steves2j, 2019-03-13:
Unfortunately I cannot share my code at the moment, it's closed IP at the moment and spread across 40+files. This issue was an observation made when trying to track down the fault. The fact that there is a loop that could potentially wait forever and no timeout or get out clause would indicate a bad design that could lead a particle in the field to be unrecoverable unless there is external intervention.

YutingYou, 2019-03-15:
@steves2j  fixed in [https://github.com/particle-iot/device-os/pull/1725](PR1725)

closed 2020-03-03 by avtolstoy