particle-iot/device-os #1737: [Gen3] USARTSerial::read() returns -260 instead of -1 for empty RX buffer

ScruffR, 2019-03-28:
Unlinke the [documentation](https://docs.particle.io/reference/device-os/firmware/xenon/#read-) states 
[image]

and also in contradiction to the known behaviour of Gen1 & Gen2 devices Gen3 returns -260 instead of -1

ScruffR, 2019-04-02:
This should be dealt with in parallel with an issue discussed with @avtolstoy  and @rickkas7  3/12/2019 about potential miscalculation of `USARTSerial::available()`.

There also is a report where code like
```
  while(Serial1.available()) Serial1.read(); 
```
seems to be blocking forever which shouldn't be the case even with permanently sending 115200 clients.

avtolstoy, 2019-04-02:
Isn't it 'blocking' simply because it instead should be `while (Serial1.available() > 0)` ?

ScruffR, 2019-04-02:
Since `Serial1.available()` should only ever render zero or positive values but *never* negative your version should be equal to `Serial1.available() != 0` which in turn again is the same as `Serial1.available() != false` which in turn again would be equal to `Serial1.available()` - at least that would be my take on the matter.

nzottmann, 2019-04-12:
I think there are deeper problems than described here, there are cases where `Serial1.available()` and `Serial1.read()` return wrong numbers. This can be positive numbers even if no data is available or wrong positive numbers if data is available.
See [this thread](https://community.particle.io/t/is-the-mesh-devices-uart-implementation-buggy/48854/34) in the Particle forum for example code for reproducing these issues.

PR #1782 (2019-05-16): [wiring] Make sure that `Serial` and `SerialX` methods are in sync with the documentation and don't return unexpected values
### Problem

#1737

### Solution

Make sure that `available()`, `availableForWrite()`, `peek()` and `read()` don't forward the errors from HAL and only return 0 or -1 in case of errors appropriately.

### Steps to Test

N/A

### Example App

N/A

### References

- Closes #1737
- [CH32372]

---

### Completeness

- [x] User is totes amazing for contributing!
- [x] Contributor has signed CLA ([Info here](https://github.com/spark/firmware/blob/develop/CONTRIBUTING.md))
- [x [... cut]
closed 2019-05-16 by sergeuz