particle-iot/device-os #1738: High power use during System.sleep for Argon

GregSawin, 2019-04-02:
## Bug Report
I was testing sleep modes on an Argon with a USB power meter and it appears to be using far more power when sleeping than not. Roughly 20mA when connected and looping and 70mA when sleeping.  The device does appear to sleep, the light goes off and checking sleepResult after shows an RTC wake.  I've confirmed this on a 2nd Argon and also confirmed this is not a problem with Xenons or Photons.

### Expected Behavior

Power usage drop to <5mA during sleep.  Not sure what the correct value is, my meter's resolution is 10mA and shows 0mA for the Photon and Xenon while sleeping.

### Observed Behavior

Power usage rises to ~70mA

### Steps to Reproduce

Flash the following to an Argon with 0.9.0 firmware and monitor power use when it goes to sleep, 10 seconds after the status led starts breathing cyan.

### Test App

```c
unsigned long lastWakeTime = 0;
void loop() {
    if (millis() - lastWakeTime > 10000) {
        System.sleep({}, {}, 60); 
        lastWakeTime = millis();
    }
 }
```

### References

https://community.particle.io/t/high-power-use-during-system-sleep-for-argon/48785
---

avtolstoy, 2020-09-23:
This should be resolved at least as of DeviceOS 2.0.0-rc.2.

Please do reopen the issue if it still persists.

GregSawin, 2020-12-04:
This is still an issue with 2.0.0.  The prior workaround to manually disable wifi still works but now requires a larger delay between the WiFi.off() and sleep commands.

@avtolstoy I don't have permissions to re-open this issue.

XuGuohui, 2020-12-04:
I'll look into this and give an update here.

GregSawin, 2020-12-04:
Some further testing shows the new sleep system does seem to work properly, no need to manually turn off Wifi.  So I'll just be using this going forward:
```
        SystemSleepConfiguration config;
        config.mode(SystemSleepMode::ULTRA_LOW_POWER).duration(60s);  //60 sec
        System.sleep(config);
```

XuGuohui, 2020-12-04:
@GregSawin Could you put your Argon into the Listening mode and use Particle CLI to check the firmware dependences. Run `particle serial inspect` and see what's echoed.

GregSawin, 2020-12-04:
Platform: 12 - Argon
Modules
  Bootloader module #0 - version 1003, main location, 49152 bytes max size
    Integrity: PASS
    Address Range: PASS
    Platform: PASS
    Dependencies: PASS
      System module #1 - version 1102
  System module #1 - version 2008, main location, 671744 bytes max size
    Integrity: PASS
    Address Range: PASS
    Platform: PASS
    Dependencies: PASS
      Bootloader module #0 - version 1003
      Radio stack module #0 - version 202
  User module #1 - version 6, main location, 131072 bytes max size
    UUID: D0DC5920EABD7EBF30BF19EEECCC67D26D35A5884C0EC0162EB229357993036E
    Integrity: PASS
    Address Range: PASS
    Platform: PASS
    Dependencies: PASS
      System module #1 - version 2008
  NCP module #0 - version 5, main location, 1536000 bytes max size
    Integrity: PASS
    Address Range: PASS
    Platform: PASS
    Dependencies: PASS
  Radio stack module #0 - version 202, main location, 192512 bytes max size
    Integrity: PASS
    Address Range: PASS
    Platform: PASS
    Dependencies: PASS

XuGuohui, 2020-12-04:
@GregSawin By using the new sleep code, did the Argon consume expected current?

closed 2020-12-04 by XuGuohui

GregSawin, 2020-12-04:
Yes, as far as I can tell.  I only have mA level resolution with my tester and it shows 0ma while sleeping vs 50-70ma with the original/classic sleep example.