particle-iot/device-os #1745: [Gen3 0.9.0] System.sleep(SLEEP_MODE_DEEP) wakes on level trigger

ScruffR, 2019-04-09:
With this code and D8 (WKP) held HIGH permanently, the device appears to enter deep sleep but immediately wakes 
```
//SYSTEM_MODE(MANUAL)
void setup() {
  pinMode(D7, OUTPUT);
  digitalWrite(D7, HIGH);
  delay(500);
}

void loop() {
  digitalWrite(D7, LOW);
  System.sleep(SLEEP_MODE_DEEP);
  // System.sleep(SLEEP_MODE_DEEP, 30);  // this should throw an error as timed wake is NOT supported - but it doesn't
}
```
Since we commonly expect a wake *only* on an edge trigger (currently `RISING` only) that behaviour is unexpected.
While Gen2 devices *never* wake in that circumstance Gen3 never really sleeps - both versions are not ideal.
While for Gen2 that appears to be an insurmountable issue with the silicon for Gen3 chances are to repair the current behaviour.

nicolaje, 2019-04-11:
I can confirm the issue on a Argon board : holding D8 high and going to deep sleep wakes up the board immediately.

It should only wake up from an edge event.

ScruffR, 2019-04-12:
Device OS 1.1.0-rc.1 has been released, can you please test with that too?

YutingYou, 2019-04-12:
On Gen3 device, nRF52840 only supports to sense high or low to exit SYSTEM OFF mode(standby mode). We use sense HIGH in Device OS.

ScruffR, 2019-04-12:
How then come the docs clearly say *"rising"* (IMO clearly suggesting and *edge* not a level)?
[image]
https://docs.particle.io/reference/device-os/firmware/xenon/#sleep-sleep-

nicolaje, 2019-04-12:
If wake on edge is not supported by the hardware, could we at least have the possibility to chose sense high / sense low when entering deep_sleep in the sleep function?

ScruffR, 2019-04-12:
AFAIK it is planned to have selectable wake pins and "edges" for deep sleep on Gen3, but preselecting the level won't solve all cases as the level may change between choosing the level and actually going to sleep.

nicolaje, 2019-04-12:
I understand, but in my case I designed my circuit to always set D8 high, unless some signal comes in which set it low, and I really counted on an edge event as documented for waking up my board.

So in my case, being able to chose sense_high or sense_low would solve my problem.

Is this something I could do at the "application" level ?

avtolstoy, 2020-03-03:
#1986 introduces configurable pin/level for Gen 3 devices for waking up from deep sleep (hibernate).

closed 2020-03-03 by avtolstoy