particle-iot/device-os #1753: Arduino compatibility: Add defines for xxxx_P() and xxxx_PF() functions

ScruffR, 2019-04-22:
There are many specialiesed Arduino functions that distinguish between PROGMEM and RAM based variables. 
See here
https://www.nongnu.org/avr-libc/user-manual/group__avr__pgmspace.html

In order to make library porting easier (or allow already contributed libraries to run - e.g. `ArduinoJson` as mentioned in [this thread](https://community.particle.io/t/arduinojson-example-fails/49234)) some "redefine" of these functions to redirect the calls to the standard functions would be helpful.

ScruffR, 2019-04-22:
Nevermind - I missed that there already are such defines for virtually every function :blush:
https://github.com/particle-iot/device-os/blob/1b1805fb4afcb93b0e92fbecfd28c082d6310285/wiring/inc/avr/pgmspace.h#L100

closed 2019-04-22 by ScruffR

shanevanj, 2019-08-14:
Including Arduino.h does not resolve the issue for a multiple file app in the WebIDE.

Inserting the following in the .h file where the ArduinoJson.h is first used provides a workaround

#ifndef strncpy_P
#define strncpy_P strncpy
#endif

#ifndef strncmp_P
#define strncmp_P strncmp
#endif

#define ARDUINOJSON_ENABLE_PROGMEM 0

This compiles 
> #ifndef strncpy_P
> #define strncpy_P strncpy
> #endif
> 
> #ifndef strncmp_P
> #define strncmp_P strncmp
> #endif
> 
> #define ARDUINOJSON_ENABLE_PROGMEM 0
> 
> #include <MQTT.h>
> #include <ArduinoJson.h>
> #include <Particle.h>
> #include "Zitron.h"
> #include "Arduino.h"

This does not

> #include <MQTT.h>
> #include <ArduinoJson.h>
> #include <Particle.h>
> #include "Zitron.h"
> #include "Arduino.h"

ScruffR, 2019-08-14:
The order of the include statements matters.
Try moving the `#include "Arduino.h"` line *before* the `#include <ArduinoJson.h>`