particle-iot/device-os #1765: [GEN3] Serial Configuration Modes

memaskal, 2019-05-06:
## Feature/Enhancement Request

GEN 3 hardware support for more Serial configurations options, such as odd parity, and 7 and 9 bit modes, which are currently not available as stated in the official Particle [documentation](https://docs.particle.io/reference/device-os/firmware/xenon)

avtolstoy, 2019-05-06:
Unfortunately this is a hardware limitation of the microcontroller used on Gen 3 devices: nRF52840. 
[image]

The only UART modes supported by the hardware are: 8N1, 8E1, 8N2 and 8E2.

We could _technically_ support 7O1, 7E1, 7O2, 7E2 by manually computing/checking the parity bit in software, so I suggest renaming the issue to something along the lines of '[GEN3] UART 7O1, 7E1, 7O2 and 7E2 modes support'.

PRs are always welcome as well :smile:

ScruffR, 2019-05-17:
If this is a feature-request 8O1 & 8O2 should not be forgotten.

avtolstoy, 2019-05-20:
> 8N2 & 8E2
I thought we had those implemented, apparently not, I've changed the title to reflect that.

> 8O1 & 8O2
We won't be able to implement these even in software, because there is no 9-bit support at all.

ScruffR, 2019-11-03:
Any news on 8N2 and 8E2?
This would help with implementing DMX512 which uses 8N2.

ScruffR, 2019-11-11:
@avtolstoy, I have also posted about the lack of 2 stop bit support on the nRF forum here
https://devzone.nordicsemi.com/f/nordic-q-a/54215/howto-use-uarte-250000-8n2-plus-txbreak-for-dmx512-protocol

and as it seems the `nrf_uarte_config()` function has not (yet) been updated to also cater for the "new" feature. 
The proposed way to go is along the line I have posted above and shouldn't be too much of a problem adding after the current call of `nrf_uarte_config()`.