particle-iot/device-os #1772: [Gen3] Multiple pinMode() issues

ScruffR, 2019-05-13:
Testing some combinations of `pinMode()` and `analogWrite()` on Argon (1.1.0-rc.1&2 & 1.2.0-beta.1) I found some enexpected behaviours

 - despite being documented as PWM capable D4 does not work with `analogWrite()`
 - after using `analogWrite()` on any pin `digitalWrite(pin, HIGH)` will never set the GPIO HIGH but LOW. 
 - D7 is by default set to OUTPUT - which is unexected as in previous generations and versions D7 was just the same as all other pins set INPUT (hi-Z) by default. This behaviour cost me some time to realise as I often use the D7 LED for quick testing whether another pin behaves as expected by jumpering to D7 but with that unexpected behaviour I suspected some issue with the tested pin before I realised this break in common practice

YutingYou, 2019-05-15:
@ScruffR  Thanks for reporting the issue!

D4 pin should be configured to OUTPUT mode before` analogWrite()`
> analogWrite() (PWM)
> ...
> NOTE: pinMode(pin, OUTPUT); is required before calling analogWrite(pin, value); or else the pin will not be initialized as a PWM output and set to the desired duty cycle.

Other issues have been fixed in https://github.com/particle-iot/device-os/pull/1777

ScruffR, 2019-05-15:
I did have that and I tested with `digitalWrite()` before actually trying `analogWrite()`.

YutingYou, 2019-05-15:
@ScruffR It worked on my side, can you please share your code?

ScruffR, 2019-05-15:
This is my code
```
const uint8_t pins[] = { A0, A1, A2, A3, D2, D3, D4, D5, D6, D8 };

int aw(const char* arg) {
  int v = atoi(arg);
  
  for (int i=0; i < sizeof(pins); i++)
    analogWrite(pins[i], v);
    
  return v;
}

int dw(const char* arg) {
  int v = atoi(arg);
  
  for (int i=0; i < sizeof(pins); i++)
    digitalWrite(pins[i], v);
    
  return v;
}

void setup()
{
  pinMode(D7, INPUT);
  
  for (int i=0; i < sizeof(pins); i++)
    pinMode(pins[i], OUTPUT);
  
  Particle.function("analogWrite" , aw);
  Particle.function("digitalWrite", dw);
}
```

When I connect D4-D7 and call `analogWrite` with a low value (e.g. 5) I see on my Argon (1.2.0-rc.1.beta) the D7 LED come on full brighness where it should be only dimmly lit.
When doing the same with any other PWM pin the expected dim D7 LED can be observerd as expected.

YutingYou, 2019-05-16:
@ScruffR  Fixed in https://github.com/particle-iot/device-os/pull/1777

avtolstoy, 2019-05-16:
I'll close this since #1777 has been merged, please reopen if necessary.

closed 2019-05-16 by avtolstoy